zmkfirmware/zmk #1420: Pure USB receiver dongle

LSChyi, 2022-08-04:
Hi, I am wondering about is there any discussion about such a feature, that creating a pure USB receiver dongle to pair with other ZMK peripherals.

The reason and usage case for this feature are:

1. The central role of the split keyboard has much less battery life, moving the receiving work to the USB dongle improves the battery life on all half.
2. Followed by 1., the peripherals can use cheaper BLE modules (e.g., NRF52832).
3. For the host that has no BLE functionality, they can insert this dongle and use keyboards wirelessly, and switching between devices is very clear as one can tell from where the USB dongle is connected to.
4. Several existing NRF52840 USB pure dongle is available on the market (e.g., [Raytac MDBT50Q-RA](https://www.raytac.com/product/ins.php?index_id=89)).

I found that the Bluetooth pairing-related implementation is kind of generic, and I did just a few changes, looks like pairing multiple peripherals is feasible, and some more work to polish such use case (e.g., disable the BLE peripheral role on the USB dongle).

Is there any discussion about this? I can help implement this feature.

caksoylar, 2022-08-05:
See #314 for related issue and #836 and #1282 on progress towards that.

LSChyi, 2022-08-05:
Good to know that, thank you @caksoylar!

closed 2022-08-05 by LSChyi