zmkfirmware/zmk #1436: LED array driver for simple indications

danricho, 2022-08-22:
Hi everyone, I'm still pretty new to ZMK so I hope I'm not duplicating.

I'm designing a shield and I'm thinking about additional features I'd like to add to it.
I'd like to add an array of LEDs (5-10 long) which could be configured in ZMK to show various things (triggered by keypress defined in the keymap).

The types of things I'd like to be able to show are:

- Battery Level (eg: 40% battery -> 11000)
- Active BLE profile (eg: 4th BLE profile active -> 00010)
- Active Layer (eg: 2nd Layer active -> 11000)

For energy preservation there would obviously be a timeout on the LEDS being lit after they are triggered.

For reference, I asked on Discord and the following examples were shared (thanks!):

- [4pplet/zmk-config-4pplet](https://github.com/4pplet/zmk-config-4pplet/blob/master/config/boards/arm/cyber60_rev_d/cyber60_led.c)
- [ergodone/zmk](https://github.com/ergodone/zmk/blob/kinesisble/app/boards/shields/kinesisble/led_worker.c)

alinelena, 2022-08-23:
i wlll add #999 to this as resource

Xoffio, 2022-09-30:
@danricho I just posted a pull request that does that #1475. if you really need it and my code does not makes it to the main branch, then you can always set your west file pointing at my branch. :)