zmkfirmware/zmk #1442: Serial API

willwade, 2022-08-31:
We’ve been working on a serial API to allow software keyboards (mainly used in the field of assistive technology eg. By people using eyegaze on a windows pc). It fundamentally is a serial interface. https://github.com/AceCentre/RelayKeys/ one of our issues is to port to esp32.. so then I found zephyr..and then zmk. Is it possible to create a serial bus to zmk and allow us to create the software interface (eg see our docs for an idea https://docs.acecentre.org.uk/products/v/relaykeys/using-relaykeys )

manna-harbour, 2022-08-31:
You could do this now using https://github.com/zmkfirmware/zmk/pull/1318 if you enable the shell over USB. But as that takes a key position index, it would mean either maintaining your keymap in both your application and ZMK, or having the full keycode set in the ZMK keymap. Better would be to add a new shell command that operates on keycodes directly.

willwade, 2022-09-03:
that looks super neat. Thanks. I'll keep a track of that. Over USB would be nice - I see there is some hacking away at that. I think this is doable - our relaykeys project could be a good demo of that functionality

Anutrix, 2023-10-28:
I am also looking for ZMK support for ESP32/ESP32-S3.

Based on https://docs.zephyrproject.org/latest/boards/xtensa/esp32s3_devkitm/doc/index.html ESP32-S3 seems to be officially supported by Zephyr so I wonder how much effort is needed to add support in ZMK.