zmkfirmware/zmk #1449: Stuck Modifier Key

duckyb, 2022-09-07:
I have been experiencing stuck modifier keys both before and after the most recent merge about this kind of issue. (I can't find the exact PR at the moment, sorry).

## So far I've found that:
+ The issue appears only when my zmk device is paired wirelessly to my PC
+ The issue appears casually, even when the keyboard is not in use
+ Pressing LCTL does not fix it, but pressing RCTL does
As a workaround I am using this macro:
```
unstick: unstick {
	label = "ZM_unstick";
	compatible = "zmk,behavior-macro";
	#binding-cells = <0>;
	bindings = <&kp LSHIFT &kp RSHIFT &kp LCTRL &kp RCTRL &kp LALT &kp RALT &kp LGUI &kp RGUI>;
};
```
+ The stuck key is either RCTL, RGUI or both

## Logs

I was able to encounter the issue during USB Logging.

I was away from my PC, and when I came back I started working on something and I pressed `LCTL + Z` to undo an action and noticed it wasn't working. Then I realized that the modifiers were stuck.

https://pastebin.com/8fix8zFV

duckyb, 2022-10-06:
Closing this since it seemed to have stopped happening. I'll update if the issue comes up again.

duckyb, 2022-11-10:
This issue keeps happening. Was scrolling reddit when all of a sudden the page zooms out. (CTRL + Scroll down)
No other keyboards are connected to my machine and I didn't touch the ZMK keyboard.

Using my "unstick" macro fixed the issue.

closed 2022-12-15 by duckyb

lovelacecreativeco, 2024-05-10:
How did you solve this? This keeps happening with mine as well

duckyb, 2024-05-10:
> How did you solve this? This keeps happening with mine as well

I'm still not sure what's been causing it. Lately it kinda stopped happening, but I think it might have been somehow caused by a Razer device I have connected.