zmkfirmware/zmk #1459: Is it possible to have a macro that remaps keys?

itsthejoker, 2022-09-14:
Hi, all! I've been looking through the docs for a bit now trying to see if this is possible, but I'm not sure if it is (beyond the "run raw C by a macro" option, which I can muscle through if I need to). I just got the new Kinesis Advantage 360 ZMK board, and what I'm trying to do is write one macro to swap LCTRL for the meta key (LGUI?), then another macro to swap them back.

If you're reading this and thinking "wtf, why", I move between Windows and Mac on the same keyboard every day. I'm used to having three separate base layouts, which I understand ZMK doesn't support, so even just swapping a few keys would solve my issues. Any help would be appreciated!

joelspadin, 2022-09-14:
ZMK does support &to layers, which effectively changes the selected layer to be the new base layer. Does that do what you want?

itsthejoker, 2022-09-14:
That sounds like something that will work! I will research that. Thank you!

closed 2022-09-14 by itsthejoker