zmkfirmware/zmk #1460: F13-F24 not functioning when NKRO enabled

ReFil, 2022-09-20:
With CONFIG_ZMK_HID_REPORT_TYPE_NKRO=y set F13-F24 don't register on the host. Commenting out that line makes the keys work with no other changes

Hello9999901, 2022-09-20:
What operating system? It is stated in the documentation that NKRO has different effects on different OSes. Personally, NKRO does not work at all on macOS.

ai212983, 2022-10-17:
@Hello9999901 Can you please point me to where exactly in the docs it is stated? [Codes page](https://zmk.dev/docs/codes#f-keys) lists F13-F24 as not supported on Android only. System configuration page [says](https://zmk.dev/docs/config/system#hid) 

> CONFIG_ZMK_HID_REPORT_TYPE_NKRO | Enable full N-key roll over. This may prevent the keyboard from working with some BIOS/UEFI versions.

Can't find anything about NKRO having "different effects on different OSes".

sr2r, 2023-06-10:
thanks for logging this issue, got it fixed after setting CONFIG_ZMK_HID_REPORT_TYPE_NKRO from `y` to `n`

ReFil, 2023-10-21:
This is the same as issue #1236, It was fixed by changing this line to use a larger HID usage https://github.com/zmkfirmware/zmk/blob/82e85699edd7261500ea2feb0697f8a6a5bbd42c/app/include/zmk/hid.h#L16

caksoylar, 2023-10-21:
@ReFil just to confirm, you are saying it works with `HID_USAGE_KEY_KEYPAD_EQUAL`, or that it works when changed to a different value? That line seems to be changed two years ago: https://github.com/zmkfirmware/zmk/commit/91ba034896c3b1668b88286e18d8f38314ec39c3

ReFil, 2023-10-21:
I had to change it to a different value, I used `HID_USAGE_KEY_KEYPAD_000` because it's greater than the f13-f24 and international Lang keys I wanted to make work

caksoylar, 2023-10-22:
Would you like to keep this open in that case, since (my understanding is) it isn't resolved?

ReFil, 2023-12-08:
Closing as this was solved in #2024

closed 2023-12-08 by ReFil