zmkfirmware/zmk #1472: Doesn't work properly with macOS karabiner-elements.

xbladesub, 2022-09-28:
I'm not sure is it 100% ZMK issue, but I can't use karabiner with it (basically can't disable built-in MacBook's keyboard).

[image]

VID and PID are not displayed some reason. Can't disable built-in keyboard while zmk board is connected. But keyboard works.

Here is my .conf:

```
CONFIG_ZMK_KEYBOARD_NAME="Chocofi"
CONFIG_USB_DEVICE_VID=0x13BA
CONFIG_USB_DEVICE_PID=0x3E7
CONFIG_ZMK_BLE=y
CONFIG_BT=y
CONFIG_ZMK_IDLE_TIMEOUT=30000
CONFIG_ZMK_SLEEP=y
CONFIG_ZMK_IDLE_SLEEP_TIMEOUT=900000
CONFIG_ZMK_MOUSE=y
```

I already reported to karabiner-elements. But as I said - I'm not sure where exactly the issue

petejohanson, 2022-09-28:
Is this when connected over USB or BT you get the results shown in the pic?

xbladesub, 2022-11-16:
It was fixed alredy

closed 2022-11-16 by xbladesub