zmkfirmware/zmk #1474: Question/Discussion: Best way to use mod morphs and combos in hold-taps?

kris-swann, 2022-09-29:
Hello, I just started playing around with zmk and just want to say how enjoyable it is to use!

I have a question on the best way to use mod morphs and combos in a hold tap behavior.

For context, here is the behavior that I am trying to achive: A combo that...
  - When tapped produces `<`
  - When shift+tapped produces `>`
  - When held produces `<=`
  - When shift+held produces `>=`

NOTE: I am using this `west.yml` so that I can use `masked_mods` in my mod morphs
```yml
manifest:
  remotes:
    - name: zmkfirmware
      url-base: https://github.com/zmkfirmware
    - name: urob
      url-base: https://github.com/urob
  projects:
    - name: zmk
      remote: urob
      revision: masked-mods
      import: app/west.yml
  self:
    path: config
```

I was able to get the behavior using this minimal `.keymap`, but was wondering if there was a better way?

```
/*
 * Copyright (c) 2020 The ZMK Contributors
 *
 * SPDX-License-Identifier: MIT
 */

#include <behaviors.dtsi>
#include <dt-bindings/zmk/keys.h>
#include <dt-bindings/zmk/bt.h>

// KEY POSITION REFERENCE
// ------------------------------------------------------------------------
// | 00 | 01 | 02 | 03 | 04 | 05 |          | 06 | 07 | 08 | 09 | 10 | 11 |
// | 12 | 13 | 14 | 15 | 16 | 17 |          | 18 | 19 | 20 | 21 | 22 | 23 |
// | 24 | 25 | 26 | 27 | 28 | 29 |          | 30 | 31 | 32 | 33 | 34 | 35 |
//                | 37 | 38 | 39 |          | 40 | 41 | 42 |

/ {

  macros {
    lte: less_than_equal_to {
      label = "macro_less_thank_equal_to";
      compatible = "zmk,behavior-macro";
      #binding-cells = <0>;
      wait-ms = <0>;
      bindings = <&kp LESS_THAN &kp EQUAL>;
    };
    gte: greater_than_equal_to {
      label = "macro_greater_than_equal_to";
      compatible = "zmk,behavior-macro";
      #binding-cells = <0>;
      wait-ms = <0>;
      bindings = <&kp GREATER_THAN &kp EQUAL>;
    };
  };

  combos {
    compatible = "zmk,combos";
    combo_abrackets_working {
      bindings = <&lk_abrackets &mm_ltgte &mm_abrackets>;
      key-positions = <6 7>;
      timeout-ms = <50>;
    };
    combo_abrackets_not_working {
      bindings = <&lk &mm_ltgte &mm_abrackets>;
      key-positions = <18 19>;
      timeout-ms = <50>;
    };
  };

  behaviors {
    mm_abrackets: mod_morph_angle_brackets {
      compatible = "zmk,behavior-mod-morph";
      label = "MOD_MORPH_ANGLE_BRACKETS";
      #binding-cells = <0>;
      bindings = <&kp LESS_THAN>, <&kp GREATER_THAN>;
      mods = <(MOD_LSFT|MOD_RSFT)>;
      masked_mods = <(MOD_LSFT|MOD_RSFT)>;
    };
    mm_ltgte: mod_morph_less_than_greater_than_equal_tos {
      compatible = "zmk,behavior-mod-morph";
      label = "MOD_MORPH_LESS_THAN_GREATER_THAN_EQUAL_TOS";
      #binding-cells = <0>;
      bindings = <&lte>, <&gte>;
      mods = <(MOD_LSFT|MOD_RSFT)>;
      masked_mods = <(MOD_LSFT|MOD_RSFT)>;
    };

    lk: linger_keys {
      compatible = "zmk,behavior-hold-tap";
      label = "LINGER_KEYS";
      #binding-cells = <2>;
      tapping-term-ms = <500>;
      quick-tap-ms = <0>;
      flavor = "tap-preferred";
      bindings = <&kp>, <&kp>;
    };
    lk_abrackets: linger_key_abrackets {
      compatible = "zmk,behavior-hold-tap";
      label = "LINGER_KEY_ABRACKETS";
      #binding-cells = <2>;
      tapping-term-ms = <500>;
      quick-tap-ms = <0>;
      flavor = "tap-preferred";
      bindings = <&mm_ltgte>, <&mm_abrackets>;
    };
  };

  

  keymap {
    compatible = "zmk,keymap";

    qwerty_layer {
      // ---------------------------------------------------------------------------------------
      // |  TAB |  Q  |  W  |  E  |  R  |  T  |           |  Y  |  U  |  I  |  O  |  P  | BKSP |
      // | CTRL |  A  |  S  |  D  |  F  |  G  |           |  H  |  J  |  K  |  L  |  ;  |  '   |
      // | SHFT |  Z  |  X  |  C  |  V  |  B  |           |  N  |  M  |  ,  |  .  |  /  | ESC  |
      //                    | GUI |  L1 | SPC |           | ENT |  L2 | ALT |
      bindings = <
        &kp TAB   &kp Q &kp W &kp E    &kp R &kp T            &kp Y   &kp U  &kp I     &kp O   &kp P    &kp BSPC
        &kp LCTRL &kp A &kp S &kp D    &kp F &kp G            &kp H   &kp J  &kp K     &kp L   &kp SEMI &kp SQT
        &kp LSHFT &kp Z &kp X &kp C    &kp V &kp B            &kp N   &kp M  &kp COMMA &kp DOT &kp FSLH &kp ESC
                              &kp LGUI &none &kp SPACE        &kp RET &none  &kp RALT
      >;
    };
  };
};

```

Specifically:
- The combo `combo_abrackets_working` gives the desired behavior
  - `<, >, >=, >=` for `tap, shift-tap, hold, shift-hold`
- The combo `combo_abrackets_not_working` gives undesired behavior
  - `h, H, g, G` for `tap, shift-tap, hold, shift-hold`
  - I assume that this is because behavior `&lk` specifies `bindings = <&kp>, <&kp>;`

Am I missing some behavior that would allow for me to specify some generic `<&modmorph>` (or even better, some generic `<&any>` that would allow for any of key presses, mod morphs, combos, etc) in the bindings that would let me use the `&lk` behavior in a generic way? Something like this?
```
    lk: linger_keys {
      compatible = "zmk,behavior-hold-tap";
      label = "LINGER_KEYS";
      #binding-cells = <2>;
      tapping-term-ms = <500>;
      quick-tap-ms = <0>;
      flavor = "tap-preferred";
      bindings = <&modmorph>, <&modmorph>;
    };
```

If this is not possible, is there a way that I don't have to provide `&mm_ltgte` ans `&mm_abrackets` in both the hold-tap definition and it's point of usage?
```
  combos {
    compatible = "zmk,combos";
    combo_abrackets_working {
      bindings = <&lk_abrackets &mm_ltgte &mm_abrackets>;
      key-positions = <6 7>;
      timeout-ms = <50>;
    };
  };
  behaviors {
    lk_abrackets: linger_key_abrackets {
      compatible = "zmk,behavior-hold-tap";
      label = "LINGER_KEY_ABRACKETS";
      #binding-cells = <2>;
      tapping-term-ms = <500>;
      quick-tap-ms = <0>;
      flavor = "tap-preferred";
      bindings = <&mm_ltgte>, <&mm_abrackets>;
    };
  }
```

Cheers, and please feel free to recommend alternative approaches if they would make more sense or lmk if this use case would warrent an update to the docs

caksoylar, 2022-09-30:
The limitations are coming from the fact that some behaviors (`#binding-cells = <1>` or `<2>` like hold-tap) except two "parameters" that modify their behavior when they are used in the keymap, whereas its `bindings` modifies the type that these parameters are passed to. This lets you not have to define `&mt A B` for each key `A B` as separate behavior instances.

In contrast, some behaviors (with `#binding-cells = <0>` like mod-morph) don't accept parameters at all, so you need to specify everything within its `bindings` property, then use the label without parameters in your keymap. As far as I know there is no way to pass other bindings as "parameters" inside a ZMK keymap layers' `bindings = < ... >;` list itself, since parameters need to be (integer) literals[^1] and not labels. (Maybe this is possible in DTS in theory but ZMK layer bindings property isn't set up to use it.)

The way you implemented it is the right way to achieve that at the moment, with one point of confusion improvement: Instead of `bindings = <&lk_abrackets &mm_ltgte &mm_abrackets>` in the combo you can use anything for the two parameters. Instead of this, you could use `bindings = <&lk_abrackets 0 0>` and it would work just fine.[^2] This is because hold-taps forward each parameter into the corresponding binding, but both hold and tap bindings are mod-morphs which don't expect a parameter, so they are ignored.

[^1]: Keycodes like `A` and `LBKT` are actually `#define`s that resolve into integers.
[^2]: Here `0` is just a dummy parameter.

kris-swann, 2022-10-01:
Amazing, thank you so much for the detailed response!

Updating `bindings = <&lk_abrackets &mm_ltgte &mm_abrackets>` to be `<&lk_abrackets 0 0>` works, is much more ergonomic, and alleviates the main pain-point I was finding!

I'll try to get a PR up to update the docs [here](https://zmk.dev/docs/behaviors/hold-tap#example-use-cases) to include a similar use-case (obviously will use an example that doesn't require use of `masked_mods`)

closed 2022-11-24 by kris-swann