zmkfirmware/zmk #1478: &caps_word gets deactivated when underscore triggered with right shift and minus

xudongzheng, 2022-10-02:
Underscore is correctly handled when using left shift with minus, however right shift with minus seems to deactivate &caps_word.

Kickeskov, 2022-10-05:
Okay I took a quick look at this.

Underlying issue: The code in `src/behaviors/behavior_caps_word.c` compares the pressed modifiers with the configured ones, matching hard on left/right. The definition for `UNDERSCORE` is defined by left shift, thus pressing RS(MINUS) does not satisfy the condition.

Quick local fix:
Change the "Continue List" in the [configuration](https://zmk.dev/docs/behaviors/caps-word) to be `<UNDERSCORE BACKSPACE DELETE (RS(ZMK_HID_USAGE(HID_USAGE_KEY, HID_USAGE_KEY_KEYBOARD_MINUS_AND_UNDERSCORE))) >`. I haven't tested it though.

Possible solutions:
1: Expand the default config to include `(RS(ZMK_HID_USAGE(HID_USAGE_KEY, HID_USAGE_KEY_KEYBOARD_MINUS_AND_UNDERSCORE)))`. (`dts/behaviors/caps_word.dtsi`)
Cons: Not pretty. Hacky. It can be added as a definition in `keys.h` but that's backwards.

2: Make the code not care about modifier sides. This can be done by doing some bitshift magic (<<4 and >>4) on the implicit/explicit modifiers in the `caps_word_is_caps_includelist` function.
Cons: There might be an edge case where this causes undesired behavior, particularly with LALT/RALT.

2b: Make the code not care about modifier sides, but only with SHIFT.
Cons: Edge case fix, where it might be better to make a general case fix by including other modifiers.

I think 2b might be the best approach. It may be possible to write the code in such a way that the bitshift magic does not seem entirely arcane, might need to introduce a helper function for this.

In order to implement the solution I need some insight from someone who knows more than me:
- Is 2b a good approach? Is it better to include all mods?
- In case the bitshift magic gets a bit too hefty and a helper function is needed, would the best place to put this be in `src/behaviors/behavior_caps_word.c`, or is there a designated place for those functions? "unified shift"?

basnijholt, 2024-01-14:
I just spend an hour scratching my head why I couldn't get this to work. Would it be worth mentioning this bug in the docs to prevent others from running into the same issue?

caksoylar, 2024-01-15:
> I just spend an hour scratching my head why I couldn't get this to work. Would it be worth mentioning this bug in the docs to prevent others from running into the same issue?

Sure, if it is worded generically (e.g. something like "keycodes that with modifier functions in `continue-list` will only be in effect if the same modifiers are used to trigger them") then that could be a good addition.