zmkfirmware/zmk #1487: Cannot connect to ZMK keyboard over Bluetooth on Pop_OS! 22.04 + Intel AX200 controller

sethforprivacy, 2022-10-10:
Hi all,

For some reason  (I believe after updating from Pop_OS! 20.10 to 22.04), I am no longer able to connect to either of my ZMK firmware keyboards over Bluetooth. I've collected all the logs I can think of below:

```txt
Bluetoothctl output:
Failed to pair: org.bluez.Error.AuthenticationFailed

dmesg output:
[ 9738.118236] Bluetooth: hci0: unexpected SMP command 0x0b from e1:ed:df:d8:a0:25
[ 9738.232020] Bluetooth: hci0: Opcode 0x 401 failed: -16

systemctl journal output:
src/device.c:load_gatt_db() No cache for E1:ED:DF:D8:A0:25

USB logging output:
[00:03:08.153,594] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[00:03:19.176,818] <dbg> zmk: connected: Connected thread: 0x20004d90
[00:03:19.176,910] <dbg> zmk: connected: Connected 44:AF:28:56:A6:C0 (public)
[00:03:19.178,405] <dbg> zmk: update_advertising: advertising from 0 to 2
[00:03:19.213,256] <inf> bas: BAS Notifications enabled
[00:03:19.300,628] <err> zmk: Security failed: 44:AF:28:56:A6:C0 (public) level 1 err 4
[00:03:19.390,991] <inf> bas: BAS Notifications disabled
[00:03:19.393,676] <dbg> zmk: disconnected: Disconnected from 44:AF:28:56:A6:C0 (public) (reason 0x05)
[00:03:19.393,707] <dbg> zmk: update_advertising: advertising from 2 to 2
```

The Bluetooth controller is the built-in controller in the Intel AX200 chipset. I've tested all things I can think of, including manually installing the latest linux-firmware, reinstalling Pop_OS! from scratch, etc.

The connection does work properly on Windows and on another Linux laptop, plus my Android phone.

https://paste.sethforprivacy.com/?d908b857102a6e0d#6Csv3EbvFSz7HeSGTNCzdJupZkze4ZxAmoqVs3fVa5Wr

Huckies, 2022-10-12:
Maybe the system upgrade wiped all the bonding informations, so the device can not be recognized anymore, and a sweep of Bluetooth profiles on keyboard may remove all the redundant bonds so they can be re-paired and re-bonded again.

sethforprivacy, 2022-10-12:
> Maybe the system upgrade wiped all the bonding informations, so the device can not be recognized anymore, and a sweep of Bluetooth profiles on keyboard may remove all the redundant bonds so they can be re-paired and re-bonded again.

The problem here is that the issue existed on two different ZMK-based keyboards, one of which was never connected before the upgrade to 22.04 as it didn't exist then.

I also cleared the first 4 profiles (the only ones I have ever tried to use on this keyboard) and that didn't resolve the issue, it was only resolved after clearing *all* profiles. Seems to be a bug of some sort that requires clearing all profiles from storage before you can properly connect in some scenarios.

19216801-Franz, 2023-03-27:
I have the same recurring issue, running ZMK on two nice!nanos for my new split keyboard. Everytime I clear the profiles / reflash the firmware the issue goes away, but returns after one successful connected session using the keyboard over Bluetooth. Any Updates on this by anybody?

Runeword, 2023-06-06:
I have the same issue on nixos 23.11 `Failed to pair: org.bluez.Error.AuthenticationFailed`

I have done all the following : Removing the device, clearing all bluetooth profiles and flashing a settings reset firmware.
After several attempts, I managed to pair the device in sudo mode.

But later I had to connect the device via usb to charge the battery, and after that the same error occur again when I try to pair the device.

Runeword, 2023-07-07:
Finally, pairing with bluetuith instead of bluetoothctl resolved the issue in my case.
https://github.com/darkhz/bluetuith

neyb, 2023-11-26:
Same issue on a glove80. Definitely a zmk strange behaviour I cannot understand (clear all only worked for me)

AlexandreBossard, 2023-11-26:
Same issue, when upgraded to kernel 6.6.2. I tried clearing all profiles and reflashing, all I get is a connect and an immediate disconnect:
```
kernel: Bluetooth: hci0: unexpected SMP command 0x0b from eb:28:2c:3b:2e:ce
bluedevil-wizard[8227]: kf.bluezqt: PendingCall Error: "Authentication Canceled"
```

neyb, 2023-11-29:
happens to me again... this time clearing all profiles did not help...

MangoIV, 2023-11-30:
I first thought this was a kernel issue (works fine on linux 6.1, which I pinned for the time being) but I'm now also using a low energy pointing device which works perfectly fine, I'm starting to think that it's at least partially the fault of ZMK that this does not work

MangoIV, 2023-11-30:
- chip: nrf52840
- kernel: 6.6.2 zen, has not been working since 6.4

[image]

MangoIV, 2023-11-30:
this is pretty serious, if ZMK doesn't support all linux kernels >6.4

petejohanson, 2023-11-30:
> I will try to do so @petejohanson do you know someone who is running a zmk keyboard on a newer kernel where it's working fine? I have had this issue on all my nrf52840 boards for a while now... I have also reported an issue in the nixos repository, some other folks also reported problems with BLE on newer kernel and wanted to take it to the kernel bug report, the issue is that now I'm using another device that is BLE and works just fine so I started thinking it's not purely the kernels fault. 

I will check my recently updated Fedora laptop to see the specific kernel version, but otherwise I've not been tracking anything at that granularity at this point. I will need to do so if we see a trend.

Can you also try my Zephyr 3.5 PR to see if anything improves?

MangoIV, 2023-11-30:
I have not updated zmk for a while, I will try this first. If issues persist, I will put logs here anyway. 

This is first running connnect, then after this fails pair. 

[le-chiff-ble-connect-6-6-zen.log](https://github.com/zmkfirmware/zmk/files/13516358/le-chiff-ble-connect-6-6-zen.log)

MangoIV, 2023-11-30:
https://hub.docker.com/layers/zmkfirmware/zmk-build-arm/3.5-branch-20231114235846-3.5.0-0.16.3-6c450465d1dd-6870957618/images/sha256-33cffa362f3613197c0c37788415271f534b80f168bb56650ba1419d2f03d46d?context=explore

I used this one...

petejohanson, 2023-11-30:
> https://hub.docker.com/layers/zmkfirmware/zmk-build-arm/3.5-branch-20231114235846-3.5.0-0.16.3-6c450465d1dd-6870957618/images/sha256-33cffa362f3613197c0c37788415271f534b80f168bb56650ba1419d2f03d46d?context=explore
> 
> I used this one... 

You'd need to actually use the ZMK branch from #1995 and `west update`. Just using the new container won't use a new Zephyr version under the hood.

MangoIV, 2023-11-30:
my github workflow runs west update; does the container not have the 3.5 zephyr? why was it named that?

petejohanson, 2023-11-30:
> my github workflow runs west update; does the container not have the 3.5 zephyr? why was it named that? 

It includes all the SDK and python dependencies needed to build something targeting Zephyr 3.5.

MangoIV, 2023-11-30:
https://github.com/NixOS/nixpkgs/issues/242313#issuecomment-1834653530

petejohanson, 2023-12-01:
> I have not updated zmk for a while, I will try this first. If issues persist, I will put logs here anyway.
> 
> This is first running connnect, then after this fails pair.
> 
> [le-chiff-ble-connect-6-6-zen.log](https://github.com/zmkfirmware/zmk/files/13516358/le-chiff-ble-connect-6-6-zen.log)

If this is logs from `btmon` then it's corrupted, perhaps.... Can we get btmon and dmesg output during the failure to connect?

MangoIV, 2023-12-01:
btmon --read works fine for me, apparently it uses some binary format when doing --write

petejohanson, 2023-12-01:
Another user reporting issues had a `dmesg` output indicating it was unhappy with the SMP security request packet... I do wonder if maybe using https://github.com/zmkfirmware/zmk/pull/2046 and enabling `CONFIG_BT_GATT_AUTO_SEC_REQ=y` will help.

petejohanson, 2023-12-01:
I also notice that the disconnects in both cases occur after the upgrade to 2M PHY... Perhaps related?

```
> HCI Event: LE Meta Event (0x3e) plen 6                                                                                                  #45 [hci0] 7.254650
      LE PHY Update Complete (0x0c)
        Status: Success (0x00)
        Handle: 16 Address: D7:4E:99:F1:BE:98 (Static)
        TX PHY: LE 2M (0x02)
        RX PHY: LE 2M (0x02)
```

petejohanson, 2023-12-01:
You also could try setting `CONFIG_BT_CTLR_PHY_2M=n` and see if that helps at all.

MangoIV, 2023-12-01:
> You also could try setting `CONFIG_BT_CTLR_PHY_2M=n` and see if that helps at all.

that worked! thank you so much!

MangoIV, 2023-12-01:
https://github.com/MangoIV/le_chiff_ble_config/pull/6

MangoIV, 2023-12-01:
@petejohanson can you perhaps enlighten us on what this setting does and if it's important for something?

petejohanson, 2023-12-01:
> @petejohanson can you perhaps enlighten us on what this setting does and if it's important for something?

It's a double data rate feature at the *lowest* level of BLE that was introduced in BT 5.0 spec. See https://www.allaboutcircuits.com/technical-articles/dive-into-the-bluetooth-low-energy-physical-layer-the-basics-of-ble-phy-radio/ for a bit more details.

For the small GATT notification sizes we are sending... It's really not critical to have enabled. If we can get further testing that indicates this improves compatibility/stability, I'll likely disable it by default in ZMK, and users can choose to then turn it back on if they want to.

I would *guess* that doing so is requiring tighter timing requirements than either the host or the ZMK keeb can adhere to and that's causing issues.

MangoIV, 2023-12-01:
Thank you for the explanation! 

If the other person (on the nixos bugtracker) also is zmk user (dunno how likely that is though) it might be present on suse as well.

I think I first experienced it on kernel 6.4, perhaps you can reproduce if you have a machine with a newer kernel?  

I have notified the other person who was affected by this on the nixos bugtracker.

Finally I can keep my kernel up to date <3

neyb, 2023-12-02:
I'm wondering if PHY M1 could have an impact on battery usage.

petejohanson, 2023-12-04:
Can everyone who is testing try another potential fix here. Instead of my previously suggested change, can you instead try setting `CONFIG_BT_AUTO_PHY_UPDATE=n`?

Doing so in theory may fix for intel, but may allow keeping 2M working for other uses, including for the split connection for wireless splits.

TIA!

neyb, 2023-12-04:
testing tomorrow if noone can before that.

neyb, 2023-12-05:
does not seems to changed anything : still tries to update to PHY 2M (then timeout on new intel firmware)

hal3e, 2023-12-05:
I have the same issue after an os update. I have a sofle clone (split keyboard) and I report that:
- `CONFIG_BT_GATT_AUTO_SEC_REQ=y` and `CONFIG_BT_AUTO_PHY_UPDATE=n` do not solve the connection problem.
- `CONFIG_BT_CTLR_PHY_2M=n` solved the connection problem but after connecting key-presses are not received on my laptop.

Is there something else I could try ?

petejohanson, 2023-12-06:
> I have the same issue after an os update. I have a sofle clone (split keyboard) and I report that:
>
>     * `CONFIG_BT_GATT_AUTO_SEC_REQ=y` and `CONFIG_BT_AUTO_PHY_UPDATE=n` do not solve the [...]

Have you completely removed and re-paired the keyboard?

wAISw, 2023-12-06:
Hi! I'm on fresh fedora linux got this problem. I have DAO keyboard. 
I solve my problem with adding
```
CONFIG_BT_CTLR_PHY_2M=n
CONFIG_BT_GATT_AUTO_SEC_REQ=y
```
https://github.com/wAISw/dao-zmk-config/commit/d30aaca69b86eedb5fb48f49b1bfe500ff6ad269

petejohanson, 2023-12-06:
> Hi! I'm on fresh fedora linux got this problem. I have DAO keyboard. I solve my problem with adding
> 
> ```
> CONFIG_BT_CTLR_PHY_2M=n
> CONFIG_BT_GATT_AUTO_SEC_REQ=y
> ```
> 
> [wAISw/dao-zmk-config@d30aaca](https://github.com/wAISw/dao-zmk-config/commit/d30aaca69b86eedb5fb48f49b1bfe500ff6ad269)

I've seen some reports of issues with the split halves connecting reliably with 2M disabled. Any issues?

hal3e, 2023-12-06:
> Have you completely removed and re-paired the keyboard?

Yes. and I thought it was ok because it connected however no key-presses were being received. 
Also, while I had the left side connected over USB the right side could not connect to the left side.

I will try the combination @wAISw mentioned and report back.

hal3e, 2023-12-06:
I have tried
```
CONFIG_BT_CTLR_PHY_2M=n
CONFIG_BT_GATT_AUTO_SEC_REQ=y
```
and everything worked!

So, I decided to try only `CONFIG_BT_CTLR_PHY_2M=n` again just to be sure. And it also works! Connection is stable for now. Both to the laptop and between halfs.

Just for reference, I did: 
- unpair keyboard from laptop
- reset both nice!nano_v2 with `settings_reset.uf2`
- flash new FW
- pair with laptop

I will monitor this thread, so if you need to test some other configuration let me know.

wAISw, 2023-12-07:
@hal3e 
I don't know what is happening, but in the morning my keyboard connects, but not type anything. I flashed again and it's now working again. I will monitor...

wAISw, 2023-12-13:
> @hal3e I don't know what is happening, but in the morning my keyboard connects, but not type anything. I flashed again and it's now working again. I will monitor...

A monitored about a week and it worked correctly.

hal3e, 2023-12-13:
> > @hal3e I don't know what is happening, but in the morning my keyboard connects, but not type anything. I flashed again and it's now working again. I will monitor...
> 
> A monitored about a week and it worked correctly.

I also confirm that after my last update the keyboard is working without any issues.

NickyTope, 2023-12-13:
same issue here with:
System 76 Oryx laptop running Arch with Intel AX211 Bluetooth chip
Custom zmk keyboard running on nice!nanos

I have tried adding the above config changes in (shield).conf as here:
https://github.com/NickyTope/zmk-config/blob/master/config/boards/shields/topey/topey.conf

None of the above let me connect to my keyboard :(

NickyTope, 2023-12-14:
Perhaps the above is my misunderstanding on where to apply these config changes for a shield type zmk config?

btmon shows that it is still doing a phy 2m update immediately before disconnect as below:

```
> HCI Event: LE Meta Event (0x3e) plen 6                                                                                                                                          #62 [hci0] 11.776859
      LE PHY Update Complete (0x0c)
        Status: Success (0x00)
        Handle: 2048 Address: EC:BC:0A:FA:E5:CA (Static)
        TX PHY: LE 2M (0x02)
        RX PHY: LE 2M (0x02)
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                                                    #63 [hci0] 11.912825
        Status: Success (0x00)
        Handle: 2048 Address: EC:BC:0A:FA:E5:CA (Static)
        Reason: Connection Timeout (0x08)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                                            {0x0001} [hci0] 11.912860
        LE Address: EC:BC:0A:FA:E5:CA (Static)
        Reason: Connection timeout (0x01)
@ MGMT Event: Command Complete (0x0001) plen 10                                                                                                                              {0x0001} [hci0] 11.912885
      Pair Device (0x0019) plen 7
        Status: Failed (0x03)
        LE Address: EC:BC:0A:FA:E5:CA (Static)
```

caksoylar, 2023-12-14:
> Perhaps the above is my misunderstanding on where to apply these config changes for a shield type zmk config?

Yes, you need to either use separate left, right conf files, or put that file under config/ directory directly: https://zmk.dev/docs/config#config-file-locations

NickyTope, 2023-12-14:
Thank you so much, can confirm the above config changes also work for me

ayatofrench, 2023-12-14:
Hello! 

I would also like to report that `CONFIG_BT_CTLR_PHY_2M=n` fixed my issue. However, it has caused one side effect in that the right half now takes much longer to connect to the left. Its not super long, but it is a noticeable wait. 

Config:
Pop-OS
Intel AX210 Bluetooth
nice!nano controllers

gyordanov, 2024-01-08:
I have the same issue and in my case:

1) CONFIG_BT_GATT_AUTO_SEC_REQ=y and CONFIG_BT_AUTO_PHY_UPDATE=n did not help
2) CONFIG_BT_CTLR_PHY_2M=n - fixed it and the left half connected to the pc, but the right half would not connect to the left half until i reset both and sometimes it will take few resets and even then it will take ... what seems like minutes to connect.

at the end i removed the changes and just kept the old bt firmware for my ax210 ... its annoying to remember to copy the old firmware (in my case ibt-0041-0041.sfi.zst)  but
better than having a working keyboard ...

vkopio, 2024-01-13:
Also confirming that `CONFIG_BT_CTLR_PHY_2M=n` fixed the connectivity issues between the left half of my board (nice!nano v2) and my laptop, but so far the right half cannot connect to the left half. There is a brief moment from time to time when the connection is successful (according to the right screen) but the connection drops immediately.

Flashing the reset firmware did not help and neither did the other proposed options.

caksoylar, 2024-01-13:
There is now an [aggregate setting](https://zmk.dev/docs/config/bluetooth) called `CONFIG_ZMK_BLE_EXPERIMENTAL_CONN` which you can set to `=y` to have a combination of the settings I shared earlier applied. Here is a copy of the call for testers that @petejohanson sent out on Discord:

>  I could use some help with testing some core changes to our BLE connection settings that are aimed to improve our reliability across all host OS/versions. You don't even need to use a different branch (assuming you are on main). For details on what the config enables, see https://zmk.dev/docs/config/bluetooth#kconfig
>
> To test:
Enable `CONFIG_ZMK_BLE_EXPERIMENTAL_CONN=y` in your config, rebuild and flash to your keeb. If using a split, please flash both sides.
> You may need to re-pair to hosts, but maybe not. If you do need to, please make a note.
> Test for a day or two.
> Report your testing experience/results in the newly created thread started with this message.
>
> Thanks in advance!

gyordanov, 2024-01-14:
@caksoylar thank you !

`CONFIG_ZMK_BLE_EXPERIMENTAL_CONN=y` seems to fix my ble issues and the two halves connect right away and seem to stay connected !

vkopio, 2024-01-14:
Looks like the `CONFIG_ZMK_BLE_EXPERIMENTAL_CONN=y` works for me as well. Only had to flash both halves and they started working immediately.

slabua, 2024-01-15:
I'm on Fedora 39 and `CONFIG_ZMK_BLE_EXPERIMENTAL_CONN` did not fix the problem.

alistairmagee, 2024-01-21:
Two computers running Debian Bookworm, 

- desktop running KDE Plasma on X11
- surface go tablet running a custom kernel form surface-linux project and Gnome Wayland

Suddenly had connection issues with the tablet when the kernel upgraded to  6.6.11. Wouldn't connect to bluetooth, secondary side would not connect to primary when connected via usb. No issues with Desktop computer (or connecting to a Mac)

using `CONFIG_ZMK_BLE_EXPERIMENTAL_CONN=y` fixed it for me.

slabua, 2024-01-21:
To those who have solved it with `CONFIG_ZMK_BLE_EXPERIMENTAL_CONN=y`,
do you also have `CONFIG_ZMK_BLE_EXPERIMENTAL_SEC=y`?

xopclabs, 2024-01-22:
I'm having this issue on NixOS (kernel 6.1.72), yesterday everything was working fine, today the ability to connect seems random. I've tried flashing with `CONFIG_ZMK_BLE_EXPERIMENTA_CONN=y`; flashing a reset firmware in prior; checkout zephyr 3.5, `west update` and build from there. No luck!

gyordanov, 2024-01-22:
i did not need `CONFIG_ZMK_BLE_EXPERIMENTAL_SEC=y`
and `CONFIG_ZMK_BLE_EXPERIMENTAL_CONN=y` is working fine with 6.7.0 and linux-firmware from few days ago (arch linux)

dilshod, 2024-01-23:
1. CONFIG_ZMK_BLE_EXPERIMENTAL_CONN=y helps fixing connecting to MacBook Pro 2017. Without this option it never connects.
2. With this option It is loosing connection to the second half split, and only flashing settings_reset helps reconnecting.

slabua, 2024-01-23:
Still no luck. And it is extremely problematic when the keyboard disconnects while typing as the last key stays stuck repeating itself for n times until it reconnects again.

alistairmagee, 2024-02-03:
> To those who have solved it with `CONFIG_ZMK_BLE_EXPERIMENTAL_CONN=y`, do you also have `CONFIG_ZMK_BLE_EXPERIMENTAL_SEC=y`?

I only used `CONFIG_ZMK_BLE_EXPERIMENTAL_CONN=y` as I read through the whole thread before attempting a fix and saw that recommended as a catch all. As it fixed the issue for me I didn't try anything else. 

What does `CONFIG_ZMK_BLE_EXPERIMENTAL_SEC=y` do? Would it stop my desktop running KDE plasma from asking me to confirm the connection when I first pair or re-pair my keyboard?

caksoylar, 2024-02-12:
> What does `CONFIG_ZMK_BLE_EXPERIMENTAL_SEC=y` do? Would it stop my desktop running KDE plasma from asking me to confirm the connection when I first pair or re-pair my keyboard?

You can check it here: https://zmk.dev/docs/config/bluetooth#kconfig 

It enables passkey entry requirement when pairing, where it asks you to enter a 6 digit pin and press enter. That sounds like what you are describing.

[19 more replies]