zmkfirmware/zmk #1488: boilerplate.cmake parses shield list vs multiple board root locations incorrectly

Percentnineteen, 2022-10-11:
While building nice_view along with an oled-enabled board that was defined in my user config area, the build process threw an error about having spi and i2c configured simultaneously. I root caused the issue to boilerplate.cmake.

Boilerplate.cmake seems to follow this basic methodology:
1. foreach board root:
2. foreach shield:
3. if shield is found, append a bunch of stuff to a bunch of lists

What happens this way is the dtsi for nice_view_adapter and nice_view are read before the dtsi for the keyboard shield

I locally modified boilerplate.cmake to fix the issue (maybe not optimally). I've attached a patch. I would have issued a pr but wasn't exactly sure how to navigate the hierarchy of git repos for this particular file.
[patch.txt](https://github.com/zmkfirmware/zmk/files/9757024/patch.txt)

caksoylar, 2023-10-18:
I think this would be pretty nice to fix, it frequently causes headaches with custom shields in user config repos, where there are various issues with the build ordering when trying to add `nice_view_adapter` and `nice_view` shields to them. There is a PR by @Percentnineteen in the Zephyr repo for it: https://github.com/zmkfirmware/zephyr/pull/24

petejohanson, 2023-10-27:
Please try again, as I've just merged zmkfirmware/zephyr#24 

Feel free to reopen if you still have any issues.

closed 2023-10-27 by petejohanson