zmkfirmware/zmk #1494: Feature request: preserve locality with tap dance

manna-harbour, 2022-10-14:
Combos do not preserve locality as noted at https://zmk.dev/docs/features/combos: 

> SOURCE-SPECIFIC BEHAVIORS ON SPLIT KEYBOARDS
Invoking a source-specific behavior such as one of the [reset behaviors](https://zmk.dev/docs/behaviors/reset) using a combo will always trigger it on the central side of the keyboard, regardless of the side that the keys corresponding to key-positions are on.

It seems that this also applies to tap dance.

A similar note should be added to the tap dance docs, or the tap dance behaviour extended to preserve locality.

bobobo1618, 2022-11-10:
Could this be resolved by setting global locality for the combo behaviours like it was for RGB underglow and ext power in https://github.com/zmkfirmware/zmk/pull/547/files#diff-fa3059ddb7abee58e50dbc5bb6b0218e6e4948d3742539e4f3a9aa67a1eb7278R149 ?

bobobo1618, 2022-11-10:
This bug also applied to global behaviours like ext power and RGB underglow for me. Without https://github.com/bobobo1618/miryoku_zmk/commit/6ba964f12a9d8138437cffacc487d4827e30fa8f, the changes only applied to the central half.

manna-harbour, 2022-11-21:
Updating this issue:

All behaviors using the locality system (https://github.com/zmkfirmware/zmk/pull/547) do not work as intended on split keyboards when the behavior is not bound directly in the keymap. If the behavior is called from any other behavior the locality information is lost and the behavior is run only on central. I.e. source-specific behaviors (https://zmk.dev/docs/behaviors/reset) invoked from indirect bindings on peripheral will run on central instead of peripheral, and global behaviors (https://zmk.dev/docs/behaviors/underglow, https://zmk.dev/docs/behaviors/power) invoked from indirect bindings will run only on central instead of on central and peripheral.

The locality system should be extended to preserve locality information when invoked indirectly.

In the mean time, the docs (https://github.com/zmkfirmware/zmk/pull/1124) should have a warning added for all behaviors with locality (pages linked above), and the current warning at https://zmk.dev/docs/features/combos should be removed. 

Work-arounds have been added to Miryoku ZMK:

- https://github.com/manna-harbour/miryoku_zmk#double-tap-boot
- https://github.com/manna-harbour/miryoku_zmk#global-shift-functions

caksoylar, 2024-09-23:
Should be supported via #2409 (except for encoder behaviors noted in #2499 and some discussion on combos in #2500), please re-open if there are still issues for nested behaviors.

closed 2024-09-23 by caksoylar