zmkfirmware/zmk #1503: Kinesis Advantage 360 Pro: can't connect bluetooth on different operating systems on the same machine

abenz1267, 2022-10-21:
Hi,

situation: 1 computer, 3 OSses (Linux, Linux, Windows). The keyboard can only pair with one OS at a time. So i basically have to clear the bluetooth profiles every time i wanna connect it to a different OS. I don't have this issue with my Logitech Lift mouse. There i can just select a different profile, pair and be done with it.

This seems like a bug to me. Is it?

Regards

joelspadin, 2022-10-21:
#549 might be relevant here?

abenz1267, 2022-10-21:
ah, so it is a design flaw with ZMK itself. that's rather unfortunate. thanks for the pointer though!!!!

closed 2022-10-21 by abenz1267

Nicell, 2022-10-21:
@abenz1267 is there any reason you can't utilize three profiles, one for each OS, so that you don't need to clear them every time you switch OSes? I don't know if I'd call this a design flaw, it's a security measure to make sure that only the exact device you paired on that profile will be the one receiving keyboard inputs.

abenz1267, 2022-10-21:
That's what i tried to do. maybe i was unclear. I'm NOT trying to use the same profile for different operating systems. I'm trying to use different profiles.

Nicell, 2022-10-21:
This is unexpected behavior. I'd suggest creating a post in our community Discord so we can ask for more details and figure out what's wrong.

petejohanson, 2022-10-21:
Yeah, that's an issue with us taking a conservative security approach, we don't allow replacing existing keys for the same BLE address.

I have some work planned for a "relaxed" security mode that would handle this, I believe more like other devices do.

MvEerd, 2022-10-22:
This is a common issue with dual booting and bluetooth pairing. Steps to reuse the same key can be found on the arch wiki; https://wiki.archlinux.org/title/Bluetooth#Dual_boot_pairing