zmkfirmware/zmk #1505: Question: How to map a key to tap for sticky layer1 and hold for momentary layer2?

pujvz, 2022-10-23:
I would like to map a thumb key to have tap for sticky layer (layer1) and hold for a different momentary layer (layer2). How would I go about doing this?

petejohanson, 2022-10-23:
Create a custom hold-tap: https://zmk.dev/docs/behaviors/hold-tap

With something like `bindings = <&mo &sl>;`

pujvz, 2022-10-23:
Thanks. Sorry if this is a dumb question. Can I put this directly into my .keymap file? 
```
/ {
    behaviors {
        mo_sl: behavior_mo_sl {
            compatible = "zmk,behavior-hold-tap";
            label = "mo_sl";
            #binding-cells = <2>;
            flavor = "hold-preferred";
            tapping-term-ms = <200>;
            bindings = <&mo>, <&sl>;
        };
    };

    keymap {
        compatible = "zmk,keymap";
        default_layer {
            bindings = <
                &mo_tog 2 1     // &mo 2 on hold, &sl 1 on tap
            >;
        };
    };
};
```

petejohanson, 2022-10-23:
> Thanks. Sorry if this is a dumb question. Can I put this directly into my .keymap file? 

That's right 

> ```
> / {
>     behaviors {
>         mo_sl: behavior_mo_sl {
>             compatible = "zmk,behavior-hold-tap";
>             label = "mo_sl";
>             #binding-cells = <2>;
>             flavor = [...]

Should be `&mo_sl` though at the end.

pujvz, 2022-10-23:
Thank you so much for pointing me in the right direction! It works exactly as described.

closed 2022-10-23 by pujvz