zmkfirmware/zmk #1506: Unable to pair split peripheral with central for XIAO BLE

cirias, 2022-10-23:
Constantly getting `<err> zmk: Security failed: D3:25:F7:E9:98:71 (random) level 1 err 9` from the usb logging.

I tested to connect the split peripheral with nRF Connect app on my phone. It worked, and I'm able to subscribe the position state.

I also tried many things below but still having the same error.

1. flash both halves to settings_reset but it's still failing.
2. add `CONFIG_ZMK_BLE_CLEAR_BONDS_ON_START=y` to [my config](https://github.com/cirias/zmk-config/blob/d761c46ac89f09688f520ed9ec9820be23bd3d14/config/8keytoy.conf) directly.
3. swap the boards of the two halves. 

According to the log, the central was able to subscribe the split service, but somehow the security failed, and it got disconnected.

```
[00:01:26.940,704] <dbg> zmk: split_central_eir_found: Found the split service
[00:01:26.941,528] <dbg> zmk: split_central_eir_found: Initiating new connnection
[00:01:27.044,036] <dbg> zmk: split_central_connected: Connected: E8:55:4C:C0:38:53 (random)
[00:01:27.044,067] <dbg> zmk: split_central_process_connection: Current security for connection: 1
[00:01:27.045,410] <dbg> zmk: split_central_process_connection: New connection params: Interval: 6, Latency: 30, PHY: 1
[00:01:27.045,410] <dbg> zmk: connected: Connected thread: 0x20004f08
[00:01:27.045,410] <dbg> zmk: connected: SKIPPING FOR ROLE 0
[00:01:27.060,974] <dbg> zmk: auth_pairing_accept: role 0, open? yes
[00:01:27.061,492] <dbg> zmk: split_central_service_discovery_func: [ATTRIBUTE] handle 27
[00:01:27.061,523] <dbg> zmk: split_central_service_discovery_func: Found split service
[00:01:27.093,048] <dbg> zmk: split_central_chrc_discovery_func: [ATTRIBUTE] handle 28
[00:01:27.093,048] <dbg> zmk: split_central_chrc_discovery_func: Found position state characteristic
[00:01:27.093,109] <dbg> zmk: split_central_subscribe: [SUBSCRIBED]
[00:01:31.125,305] <dbg> zmk: split_central_notify_func: [UNSUBSCRIBED]
[00:01:31.125,335] <dbg> zmk: split_central_chrc_discovery_func: Discover complete
[00:01:31.126,251] <err> zmk: Security failed: E8:55:4C:C0:38:53 (random) level 1 err 9
[00:01:31.127,136] <dbg> zmk: split_central_disconnected: Disconnected: E8:55:4C:C0:38:53 (random) (reason 8)
[00:01:31.127,136] <dbg> zmk: release_peripheral_slot: Releasing peripheral slot at 0
[00:01:31.127,471] <dbg> zmk: start_scan: Scanning successfully started
```

snsten, 2022-10-26:
I had the same issue here's what worked in my case.
Create a keymap with a key mapped to &bt BT_CLR and try to press it on peripheral.

If that doesn't work then do the following:
1. Flash the central with settings_reset, flash firmware for the left side and disconnect from power.
2. Flash the peripheral with settings_reset, flash firmware for the right side and disconnect from power.
3. Connect first the central and then peripheral side.
4. If it still doesn't connect press the key mapped to &bt BT_CLR on peripheral side and it should work.

cirias, 2022-10-27:
It ordered a new board. And finally they worked! Probably something weird happened to one of the boards.

closed 2022-10-27 by cirias

caksoylar, 2022-11-01:
One point of clarification for future readers: `&bt BT_CLR` or other `&bt` behaviors don't affect peripheral/central pairing at all. They are strictly for central-to-computer connections.