zmkfirmware/zmk #1517: Bug? LA(RIGHT) !== Alt + RightArrow

arabshapt, 2022-10-30:
I have a shortcut for "go back"/"go forward" in browser. LA(RIGHT or RIGHT_ARROW) on certain pages(e.g. theverge.com) focuses on three dots of chrome browser and MS Edge (menu button) as though I clicked f10. This is happening on windows 10 and 11. I didn't test on  any other OS. I am using Kinesis 360 Pro. Is there any other way to do Alt + Arrow other than LA(Right)?

Just hitting alt+(right or left) never has this behavior and works as expected.

auipga, 2022-11-29:
> as though I clicked f10 

The Alt sets the focus there too if no other key is pressed iirc.

Maybe you try the keycodes K_BACK and K_FORWARD.

arabshapt, 2022-12-03:
Thank you for your suggestion) I will try that. Yes, Alt on its own does that, but LA(RIGHT) should(or I expected) not set focus anywhere... When I do it manually with two keys on my keyboard, it doesn't change focus. At the moment I am using hyperkey(left_arrow) and I remap it to Alt Left in windows powertoys and works great, but of course it would be better to be able to do it without external apps.

auipga, 2022-12-05:
Just a guess: Did you wrap &kp LA(RIGHT) into something else like a macro or encoder?

Eryx5502, 2022-12-06:
I'm having an issue which might be the same as this one. Or at least there's some coincidence. I have a shortcut `&kp LG(RIGHT)` which, after producing a `WIN + RIGHT` combination, it opens windows menu. I guess it's triggering an extra `WIN` press after the intended key combination.

The same happens with all other arrows instead of `RIGHT`. I have no clue on what could be happening (I'm fairly new to ZMK), but I did try usb logging which I could upload if you think it is worth it.

jangraefen, 2025-05-31:
I am also facing the behavior while binding LG(<any-arrow-key>). It always fires an extra GUI. I cannot observe the same, binding other keys with the LG modifier function.