zmkfirmware/zmk #1519: ModTap SHIFT + ENTER does not work

mateuszaaa, 2022-10-31:
hi guys, im using zmk firmaware for my kyria, for some reason i cannot get to work my setup, i would like to use a key as an shift when pressed and as an enter when quickly typed. 

Same thing works for me when using LShift + esc
```
&mt LSHIFT ESC
```

but it doesnt when used with enter
```
&mt RSHIFT ENTER
```

my config for reference

https://github.com/mateuszaaa/zmk-config/blob/master/config/kyria_rev2.keymap

caksoylar, 2022-11-01:
There is no reason it shouldn't; it has been working OK for me for a very long time, e.g. at https://github.com/caksoylar/zmk-config/blob/main/config/3x5.keymap#L112. Maybe the specific application you are using isn't working well with Enter key's tap duration? This happens for some programs, since ZMK sends a tap event of a fixed duration for hold-tap taps but they expect a longer duration.

mateuszaaa, 2022-11-02:
im just using vim, it doesnt work in terminal as well ;/

caksoylar, 2022-11-02:
Are you using Gnome Terminal?

mateuszaaa, 2022-11-02:
i checked in gnome-terminal, xterm & termite

caksoylar, 2022-11-02:
It is at least a known issue in some Gnome environments https://github.com/zmkfirmware/zmk/issues/759#issuecomment-1193182843 and Gnome terminal (from some reports on Discord). In any case, sounds like it might be a duplicate of #759 .

mateuszaaa, 2022-11-07:
that super weird but reflashing keyboard few times solved the issue, had similiar problems with

```
&lt SPACE NUMLAYER
```

reflashing few times both boards helped -.-