zmkfirmware/zmk #1523: Doc: Programmer Dvorak implementation

taqtiqa-mark, 2022-11-06:
Is it possible to implement [programmer dvorak](https://www.kaufmann.no/roland/dvorak/) layout without installing drivers etc on windows, macos and linux?

If I understand correctly this should be able to be implemented via a keymap or a behavior?  Doubts arise over the inversion of the numbers and symbols.

Finally, if this is all possible is aPR to this project appropriate, or do these layouts belong in their own repo?

Appreciate any hints or tips.

caksoylar, 2022-11-06:
I might not be fully understanding how to activate the alternative functions of the keys from the explanation webpage. However in its most basic form I believe this layout assigns certain modifier+key combinations to produce different keycodes, such as `shift+7` resulting in `[`. You can replicate this approach using the [mod-morph behaviors](https://zmk.dev/docs/behaviors/mod-morph).

Having said that, using custom keyboard firmware allows you to define multiple [layers](https://zmk.dev/docs/features/keymaps#layers) that can be activated via dedicated keys, and don't depend on overloading modifier functionality. Most firmware users prefer using those to add additional functionality to key positions and I'd strongly encourage doing that as well. At the end of the day this layout seems to be designed considering the limitations of non-programmable keyboards.

Note that for some of the symbols you would also need to use [macros](https://zmk.dev/docs/behaviors/macros) to send a sequence of keycodes to produce the desired symbol. The specific sequence will be dependent on the OS, see [one example](https://zmk.dev/docs/behaviors/macros#unicode-sequences) in the docs and general discussion in #232.

To get the dead keys functionality you would likely need to use an OS layout that uses dead keys, like US International.

Greek and math mode would be possible by toggling on a special layer that contains macros to produce the symbols in unicode.

I would encourage looking through other people's ZMK configs (some of which are pretty well documented) through [the Github topic](https://github.com/topics/zmk-config) for inspiration. Some of us are using keyboards as small as 30 keys, where we can cram all the necessary functionality through the use of layers and combos.

caksoylar, 2023-12-11:
I am closing this from ZMK point-of-view, since we aren't likely to get a section for this in the docs. The recipes section proposed in https://github.com/zmkfirmware/zmk/pull/1953 might be a good fit if anyone is inclined to contribute an item on programmer Dvorak.

closed 2023-12-11 by caksoylar