zmkfirmware/zmk #1526: Feature requests: swap hands for split keyboards

mike1808, 2022-11-08:
## Background 

They are cases when you need type one-handed without moving your hand to the other split but be able to press the keys on it. That's especially relevant when the splits are far apart from each other or when the pointing device is in-between the splits. For instance, when working in CAD software many times you required to hold one key while moving your mouse, if you use your mouse with left hand you will need to reach your right hands from the left side to the right split. 

To mitigate that, QMK has [swap hands](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_swap_hands.md) feature s which when enabled it maps the key presses of one side to the other. 

## Why

Introducing swap hands feature will eliminate excessive hand movement and also remove the need of defining an extra layer in the keymap.
 
I wanted to create an issue to track the progress and discuss it.

yangit, 2023-06-20:
I think it belongs to config generators.
There are many ways you can implement it, I use my generator to create
- mirror letters
- mirror symbols
- mirror arrows
- mirror shift layer

And some of the mirrors require custom adjustments to stay sane.

I do not think it belongs to ZMK config any more than sane tap dance declaration and a lot of other things which are not easy to define with ZMK syntax
You can look [here](https://github.com/yangit/zmk-config/blob/main/yanConfig/index.ts#L142) for how I define mirror layers in my config generator, but it is written in JS so yeah, not part of ZMK.

mike1808, 2023-06-20:
@yangit I do think this belongs to the firmware. The firmware can define the mapping definition on the physical matrix layer which then can be used in the keymap.

Yes, you can work around and make it work with generators or other tools but those are workarounds requiring a lot of knowledge or manual work and not solutions.

I did implemented my swap layer using C macro https://github.com/mike1808/zmk-config/blob/main/config/helpers.h#L22 for now.

kovasap, 2024-08-01:
+1 for this feature