zmkfirmware/zmk #1534: hold tap: tap behavior triggers to fast for Ubuntu with modifiers

tyalie, 2022-11-11:
When using any of the hold tap behaviours and having the tap behaviour be combined with a modifier, Ubuntu will struggle to register the latter.

The issue seems to be that Ubuntu (using libinput and wayland) struggles or maybe even on purpose ignores the modifier key if it was pressed for a too short amount of time. ZMK hits that limit when using a tap in tap-hold behaviour as it can result in a difference of 1ms between press and release as shown by `evtest`.
```
Event: time 1668188704.385420, type 4 (EV_MSC), code 4 (MSC_SCAN), value 700e5
Event: time 1668188704.385420, type 1 (EV_KEY), code 54 (KEY_RIGHTSHIFT), value 1
Event: time 1668188704.385420, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70006
Event: time 1668188704.385420, type 1 (EV_KEY), code 46 (KEY_C), value 1
Event: time 1668188704.385420, -------------- SYN_REPORT ------------
Event: time 1668188704.386408, type 4 (EV_MSC), code 4 (MSC_SCAN), value 700e5
Event: time 1668188704.386408, type 1 (EV_KEY), code 54 (KEY_RIGHTSHIFT), value 0
Event: time 1668188704.386408, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70006
Event: time 1668188704.386408, type 1 (EV_KEY), code 46 (KEY_C), value 0
Event: time 1668188704.386408, -------------- SYN_REPORT ------------
```

I've added a 20ms delay `behavior_hold_tap.c:release_binding` to circumvent this and it works flawlessly. Normal human tapping behaviour seems to be around 50ms, maybe ZMK should orient along that line.

caksoylar, 2022-11-11:
Related: #759

caksoylar, 2023-12-11:
For posterity, this seems to be a duplicate of #759 since it has become clear that Gnome with Wayland has this issue which should be fixed by Gnome 45 release.

closed 2023-12-11 by caksoylar