zmkfirmware/zmk #1538: Not being able to connect keyboard to computer, NixOS

Eken-beep, 2022-11-14:
Trying to connect the keyboard to my computer which bluetooth works on normally but not for this keyboard. The keyboard doesn't connect to a windows laptop either or my iphone.
The keboard connects successfully but directly after it disconnects with no hints given however dmesg provides the following output:
`[16115.648227] Bluetooth: hci0: unexpected SMP command 0x0b from e0:6b:f7:16:0a:01
[16122.848229] Bluetooth: hci0: unexpected SMP command 0x0b from e0:6b:f7:16:0a:01
[16124.598219] Bluetooth: hci0: unexpected SMP command 0x0b from e0:6b:f7:16:0a:01`
I use a custom shield which is on my github: https://github.com/Eken-beep/zmk-config

closed 2022-11-21 by Eken-beep

markhv-code, 2023-04-21:
I have a similar issue with my Kinesis Advantage 360 Pro keyboard. I'm on PopOS 22.10

bendlas, 2023-08-03:
I also have this issue with my corne + nice!nano

aitvann, 2023-08-18:
similar issue with my dao + nice!nano on NixOS. clearing all profiles and resetting did not fixed the issue although I am able to connect the keyboard to an android phone with no problem

billksun, 2023-09-06:
I just ran into this issue myself with my new Corne + nice!nano combo. In case someone finds this page in the future, here's how I solved it.

nice!nano is a Bluetooth Low Energy device, so some Linux Desktops may have issues scanning for them. Follow the instructions in this ArchWiki article: https://wiki.archlinux.org/title/Bluetooth#Device_does_not_show_up_in_scan

Using `bluetoothctl` and changing the scan transport to LE was enough for my system to detect the keyboard.

woodentoken, 2023-09-06:
I'm having the same issue on linux mint with a nice!nano v2. I'm getting the following errors in the following places: 

bluetoothctl: `org.bluez.Error.AuthenticationFailed` error when attempting to 'pair'

syslog: `src/device.c:load_gatt_db() No cache for <device MAC address>`

dmesg: `Bluetooth: unexpected SMP command 0x0b from <device MAC address>`

these all seem related, but which is the origin? not sure yet.

@billksun If i understand correctly the issue is about connecting to the device, not finding it in the first place. I have no issue locating the device. Were you having trouble *locating* the device or actually connecting to it?

billksun, 2023-09-06:
@woodentoken, you are correct, my problem was the keyboard not showing up when scanning for devices to pair. Once the keyboard was detected, I had no issues connecting. Again, I believe my issue is specifically with BLE devices.

For what it's worth, I am using KDE Plasma 5.27 on NixOS.

woodentoken, 2023-09-06:
@billksun all good, just wanted to ensure we're on the same page

I actually got a stable connection by clearing the "current" Bluetooth profile that was on the nice!nano (surely the board had previously been connected to another computer on this profile - maybe a screen would have been a good idea in retrospect).

Was able to connect to the board after doing that - the existing bluetooth profile was probably blocking the connection.

Runeword, 2023-09-10:
> similar issue with Dao keyboard on NixOS. clearing all profiles and resetting did not fixed the issue although I am able to connect the keyboard to an android phone with no problem

I had the same problem on NixOS.  Clearing all profiles + using https://github.com/darkhz/bluetuith instead of bluetoothctl worked.