zmkfirmware/zmk #1546: Building from zmk-config folder throwing cmake pathing issue on windows

razlani, 2022-11-19:
As per title, [Building from zmk-config Folder](https://zmk.dev/docs/development/build-flash#building-from-zmk-config-folder) in windows is throwing a cmake issue whereby it expects unix-style path (presumably from the error):

    Syntax error in cmake code at

    C:/Users/privy/git/keebs/zmk/zephyr/cmake/dts.cmake:2147483647

    when parsing string

    C:\Users\privy\git\keebs\miryoku_zmk\config/corne.keymap

    Invalid character escape '\U'.

Full error log enclosed below

    $ west build -b nice_nano -- -DSHIELD=corne_left -DZMK_CONFIG="C:\Users\privy\git\keebs\miryoku_zmk\config"
    -- west build: generating a build system
    Including boilerplate (Zephyr base (cached)): C:/Users/privy/git/keebs/zmk/zephyr/cmake/app/boilerplate.cmake
    -- Application: C:/Users/privy/git/keebs/zmk/app
    -- Adding C:/Users/privy/git/keebs/zmk/app/boards/shields/corne
    -- ZMK Config directory: C:\Users\privy\git\keebs\miryoku_zmk\config
    -- Using keymap file: C:\Users\privy\git\keebs\miryoku_zmk\config/corne.keymap
    -- Zephyr version: 3.0.0 (C:/Users/privy/git/keebs/zmk/zephyr)
    -- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
    -- Board: nice_nano, Shield(s): corne_left
    -- Cache files will be written to: C:/Users/privy/git/keebs/zmk/zephyr/.cache
    -- Found toolchain: gnuarmemb (C:/12.2-mpacbti-bet1)
    CMake Error at C:/Users/privy/git/keebs/zmk/zephyr/cmake/dts.cmake:2147483647 (zephyr_variable_set_too_late):
    Syntax error in cmake code at
    C:/Users/privy/git/keebs/zmk/zephyr/cmake/dts.cmake:2147483647
    when parsing string
    C:\Users\privy\git\keebs\miryoku_zmk\config/corne.keymap
    Invalid character escape '\U'.
    Call Stack (most recent call first):
    C:/Users/privy/git/keebs/zmk/zephyr/cmake/dts.cmake:70 (if)
    C:/Users/privy/git/keebs/zmk/zephyr/cmake/app/boilerplate.cmake:542 (include)
    C:/Users/privy/git/keebs/zmk/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
    C:/Users/privy/git/keebs/zmk/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
    CMakeLists.txt:15 (find_package)
    CMake Error: Error in cmake code at
    Unknown:0:
    A command failed during the invocation of callback "zephyr_variable_set_too_late".
    -- Configuring incomplete, errors occurred!
    FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' '- 
    DWEST_PYTHON=C:\Python311\python.exe' '-BC:\Users\privy\git\keebs\zmk\app\build' '-SC:\Users\privy\git\keebs\zmk\app' - 
    GNinja -DSHIELD=corne_left '-DZMK_CONFIG=C:\Users\privy\git\keebs\miryoku_zmk\config'

Searching issues and bafflingly I couldn't find any open ones - is it something I'm doing?

joelspadin, 2022-11-19:
Try forward slashes instead of backslashes. It appears to be parsing the backslashes as escape characters, so it sees `C:\Users` as `C:` followed by the character `\U`, which isn't a valid escape character.

razlani, 2022-11-19:
> Try forward slashes instead of backslashes. It appears to be parsing the backslashes as escape characters, so it sees `C:\Users` as `C:` followed by the character `\U`, which isn't a valid escape character.

Uhhh I realized that my editor uses bash by default so that may have been confusing things (whoopsy). 

So I try a 
    west build -b nice_nano -- -DSHIELD=corne_left -DZMK_CONFIG="/c/Users/privy/git/keebs/miryoku_zmk/config"
in bash and same error

So I opened up regular ole' cmd.exe and executed a 

    west build -b nice_nano -- -DSHIELD=corne_left -DZMK_CONFIG="C:/Users/privy/git/keebs/miryoku_zmk/config"

Same error.

Sooo it looks like it's broken and not responding to input at this point?

I try using a joke config of 

    west build -b nice_nano -- -DSHIELD=corne_left -DZMK_CONFIG="LOLCATS"

Same error.

I delete cache randomly from a .cache folder, thinking maybe it's caching the last error command? 

Nada!!

caksoylar, 2022-11-19:
If you are not building with the `-p/--pristine` switch it will probably be caching the previous config in the build directory itself; to be certain remove the build directory before building again. I'd also try `-DZMK_CONFIG="C:/Users/privy/git/keebs/miryoku_zmk/config"`, in case you haven't yet.

razlani, 2022-11-21:
> If you are not building with the `-p/--pristine` switch it will probably be caching the previous config in the build directory itself; to be certain remove the build directory before building again. I'd also try `-DZMK_CONFIG="C:/Users/privy/git/keebs/miryoku_zmk/config"`, in case you haven't yet.

pristine was the magic flag in conjunction with using unix style paths within the bash shell.. So PEBCAK in a nutshell for this one I thinks, closing up thanks gents.

closed 2022-11-21 by razlani