zmkfirmware/zmk #1559: IO error when trying to flash nice!nano2

alex8065, 2022-11-30:
```[18036.977115] sd 2:0:0:0: [sdc] Attached SCSI removable disk
[18062.856072] usb 1-4: USB disconnect, device number 24
[18062.857649] blk_update_request: I/O error, dev sdc, sector 517 op 0x1:(WRITE) flags 0x100000 phys_seg 1 prio class 0
[18062.857673] Buffer I/O error on dev sdc, logical block 517, lost async page write
```

This happens consistently, and every time sector 517.

717a56e1, 2022-12-01:
2 computers (both running Linux, but different distributions and kernels), various USB cables. It's happening on both nice!nano's I bought, even the one I flashed successfully before. These particular messages might be a red herring, but I still can't flash the right nice!nano regardless.

alex8065, 2022-12-01:
2 computers (both running Linux, but different distributions and kernels), various USB cables. It's happening on both nice!nano's I bought, even the one I flashed successfully before. These particular messages might be a red herring, but I still can't flash the right nice!nano regardless.

zaddok, 2022-12-06:
On mac, this is a finder error, you bypass it by copying using the terminal, i.e.

`cp corne_left-nice_nano_v2-zmk.uf2 /Volumes/NICENANO/`

nmunnich, 2025-08-15:
Issue seems to have been resolved.

closed 2025-08-15 by nmunnich