zmkfirmware/zmk #1569: docs: Setup instructions for macOS on Apple silicon

dxmh, 2022-12-04:
The `gcc-arm-embedded` requirement for ZMK/Zephyr 3.0 on macOS seems to be available for x86_64 host architecture only ([source](https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads)). 

According to a quick search on the ZMK Discord server, Apple silicon users aren't having much luck running it via Rosetta.

It would be good to document this with the [MacOS setup instructions](https://github.com/zmkfirmware/zmk/blob/001105e31844f0f59b82a2e71a4c3ada980b931b/docs/docs/development/setup.md#gnu-arm-embedded-1) and possibly suggest an alternative (such as Docker (https://github.com/zmkfirmware/zmk/pull/879), or the [Zephyr SDK for macOS](https://github.com/zephyrproject-rtos/sdk-ng#releases)).

Hello9999901, 2022-12-04:
I have a PR for something similar right now. I can add this in with more verbose detail.

caksoylar, 2023-11-05:
This is now covered by #1621, which links to Zephyr docs that includes specific instructions for Apple Silicon.

closed 2023-11-05 by caksoylar