zmkfirmware/zmk #1580: board_index in setup.sh is null

Kadenias, 2022-12-11:
Hi folks,

i just tried to flash my nice!nanos for a new corne keyboard using the config setup script and was getting the error:
**"Wired split not yet supported"**

I choose corne in the keyboard selection with "13" and nice!nanos v2 with "6". 

So I did a bit of digging in the setup.sh and it seems the variable board_index is initialized after it is used to get the corresponding value from boards_usb_only array:

```
                if [ -n "${!keyboard_sibling_first}" ] && [ "${boards_usb_only[$board_index]}" = "y" ] ; then
                    echo "Wired split is not yet supported by ZMK."
                    exit 1
                fi
```

petejohanson, 2022-12-11:
@Kadenias Indeed, good catch. We were getting lucky previously because the first item in the list was always a wireless board. Moved the check in #1581, seems to be behaving now, please try again in a few minutes once Netlify deploys.

caksoylar, 2023-12-11:
Closing since this should be fixed now, cannot reproduce.

closed 2023-12-11 by caksoylar