zmkfirmware/zmk #1583: nibble underglow and nice!nano v2 support

jhsu, 2022-12-13:
When trying to enable underlgow with nibble board + nice!nano v2, i'm getting a build error as it can't determine the number of leds: https://github.com/jhsu/zmk-config/actions/runs/3681553747/jobs/6228473449

shouldn't "underglow" be listed in features? https://github.com/zmkfirmware/zmk/blob/f03e29ca135fb41644cbae40b9233ba7adce4153/app/boards/shields/nibble/nibble.zmk.yml

caksoylar, 2022-12-13:
Looks like this is the same issue as https://github.com/zmkfirmware/zmk/issues/885. You can try the workarounds in the issue such as copying [the overlay](https://github.com/zmkfirmware/zmk/blob/main/app/boards/shields/nibble/boards/nice_nano.overlay) into your keymap or `#include` it like `#include <../boards/shields/nibble/boards/nice_nano.overlay>`.

jhsu, 2022-12-13:
yep! that seems to have been the issue. closing as this is a duplicate.