zmkfirmware/zmk #1588: Tap dance and hold tap behaviour

kriku, 2022-12-14:
Hello!

So apparently, if I set tap-dance behaviour on some key, and trying use this button with shift modifier, I have to wait `tapping-term-ms` and only after it I can release shift modifier.

For example:

I set some key to         
```
        tdz: tap_dance_0 {
            compatible = "zmk,behavior-tap-dance";
            label = "TAP_DANCE_0";
            #binding-cells = <0>;
            tapping-term-ms = <215>;
            bindings = <&kp SEMI>, <&kp GRAVE>;
        };
```

It works as expected, double tap emits `` ` ``. 

But when I trying to type ` : `, I have to release shift modifier strictly after tapping timeout otherwise it emits ` ; `, because shift modifier is already released.

As I see it:
```
|shift-------|
^ firstly I press shift
             ^ I release shift before tap-dance bindings emit a symbol
....|;------------|
    ^ I press semicolon and tapping timeout starts
                  ^ after tapping timeout ends - emit symbol, but shift already released

```

Is there any way to interrupt tap waiting on releasing of other key? Actually right before release? To fix this case with in particularly

yanshay, 2023-03-10:
I also stumbled on a few tap-dance issues so trying to understand the code. From what I understand, tap-dance invokes the bindings either on tapping-term-timeout-ms or in case it is interrupted by another key. However it doesn't consider a release as interruption, it seems to ignore 'upstroke events'. So in your case it gives a chance to the second tap to be pressed and when it reaches the timeout the shift is no longer pressed.

I think it would make sense for it to also terminate on certain key release (like mods). You can try to check this out by commenting out ignoring upstroke events. I don't know what other implications this may have, but worth a try. If it works can add configuration of which key upstrokes not to ignore.

```
static int tap_dance_position_state_changed_listener(const zmk_event_t *eh) {
    struct zmk_position_state_changed *ev = as_zmk_position_state_changed(eh);
    if (ev == NULL) {
        return ZMK_EV_EVENT_BUBBLE;
    }
    if (!ev->state) { // <-- comment this section out
        LOG_DBG("Ignore upstroke at position %d.", ev->position);
        return ZMK_EV_EVENT_BUBBLE;
    }
    for (int i = 0; i < ZMK_BHV_TAP_DANCE_MAX_HELD; i++) {
    .
    .
    .
```

towe75, 2023-10-27:
I can confirm the initial problem and tried @yanshay's hack. It fixed the problem for me.
Can some zmk developer comment if this is a appropriate solution? Putting together a PR with a new config option should be straight forward.