zmkfirmware/zmk #1589: Possible to have Alt-Tab switching on a Function Key?

drelephant, 2022-12-14:
This might be a question more than an issue...

Is it possible to have a layer switch key to act as the Alt key when held in ZMK?

ie so I can have a momentary layer key (for my movement layer with arrow keys/pgup/pgdn/etc), but also have keys on that layer to do Alt-Tab and Alt-Shift-Tab if you keep the layer key held.

eg see [this QMK commit](https://github.com/davidagross/qmk_firmware/commit/63208a37c2b88a739591608b85e1fb794406235d#diff-837a0e3a3a20c83ec8d3ace09f5b8d97) - it has a variable `is_alt_tab_active` that will "hold" it in the layer if you keep the layer key held.

caksoylar, 2022-12-14:
See https://github.com/zmkfirmware/zmk/issues/997#issuecomment-1314549032, it is currently possible with a PR.

drelephant, 2022-12-14:
Thanks for pointing me to that. That's good that it seems possible, because it's something I use quite often.

Hmm, seems pretty complicated, I'm completely new to ZMK and just about to try with an nrf52840... Hopefully I'll understand it once I get started.

caksoylar, 2022-12-15:
Feel free to ask for help once you get started, I and others would be happy to help over at the [official Discord server](https://zmk.dev/community/discord/invite).

drelephant, 2022-12-15:
Thanks very much, still designing the hardware but soon!

closed 2022-12-15 by drelephant