zmkfirmware/zmk #1599: feat: add option to disable storing state to settings

xudongzheng, 2022-12-20:
Currently ZMK uses settings to periodically remember the latest state for `&out`, `&ext_power`, and others.

Some users may be prefer for such state to not persist. Not using settings means you are guaranteed a consistent state after each reset. The 60 second default debounce period can sometimes make it difficult to know what the initial state is.

One approach is setting `CONFIG_ZMK_SETTINGS_SAVE_DEBOUNCE` to a really high value. However this is imperfect as setting that already exists on storage will persist indefinitely.

Ideally there would be a special value (such as `CONFIG_ZMK_SETTINGS_SAVE_DEBOUNCE=-1`) or a separate Kconfig value that disables state from being read from settings (and uses the default state). Instead of just `IS_ENABLED(CONFIG_SETTINGS)`, it would check the proposed setting as well https://github.com/zmkfirmware/zmk/blob/main/app/src/ext_power_generic.c#L144.

joelspadin, 2022-12-20:
I'm not sure I understand this. How would this be different from disabling `CONFIG_SETTINGS`?

petejohanson, 2022-12-20:
> I'm not sure I understand this. How would this be different from disabling `CONFIG_SETTINGS`?

Unfortunately, this would also break BT key storage... So we would need something more nuanced.

Timoyoungster, 2024-04-16:
Wouldn't it be enough to change this ```CONFIG_SETTINGS``` (line 44) to a new config value?

https://github.com/zmkfirmware/zmk/blob/e22bc7620cef763d9ad80e9b98182273de9973db/app/src/endpoints.c#L43-L49

If that's correct I'd also be happy to create a PR.
*(would be my first one though, so it might be a pretty bad one)*

Timoyoungster, 2024-04-16:
Ok, clear. But it won't break any other functionality this way right?

Timoyoungster, 2024-05-30:
If someone else is still waiting for this, I created a PR (#2308) 
and would be happy if also other people could test it. (I am only able to verify on one single keyboard)