zmkfirmware/zmk #1608: External Power Control toggle/turn off not respected by right side of split

razlani, 2022-12-23:
So I'm [reading online](https://zmk.dev/docs/behaviors/power#split-keyboards) that externally turning off power affects both halves of the split equally.

Cool not super nuanced but I can live with that: my goal is to have both LCDs connected at all times, but just flash them on to see if they need a charge using the toggle shortcut (and then resetting them both, given an ongoing zmk issue preventing lcd on after a 'sleep').

Issue is that only the left (and main) side seems to respect the toggle.

The only thing I can do to get the right side to turn off is use a hack of setting a very low figure of

`CONFIG_ZMK_IDLE_TIMEOUT`
As [per this](https://zmk.dev/docs/config/power). Right side of split seems to respect this setting, but it's a crap solution as my right side battery will still take a beating.

Any one know of how I can troubleshoot this issue, assuming it is one and not just PEBCAK?

Noteworthy that I'm using 2x n!n (version 2) and miryoku-zmk for config.

Flashing via `west`, 
`west build -p -b nice_nano_v2 -d build/right -- -DSHIELD=corne_right -DZMK_CONFIG="/mnt/c/Users/privy/git/keebs/miryoku_zmk/config"`

 `West version: v0.14.0`

Last ZMK commit 
```
commit 70aa4fa82456ab6a739c43aae36815804cb58d51
Author: Kyle McCreery <mccreery.kyle@gmail.com>
Date:   Sat Dec 10 23:01:37 2022 -0500
```
And then merging mouse-ftc on top.

caksoylar, 2022-12-23:
You might be running into the #1494. Miryoku was updated recently to not rely on mod morphs for these by default, see https://github.com/manna-harbour/miryoku_zmk/#global-shift-functions.

caksoylar, 2023-12-11:
Closing this as a duplicate of #1494, without a response.

closed 2023-12-11 by caksoylar