zmkfirmware/zmk #1620: GPIO LED PWM controlled by behaviors?

idesignstuff, 2023-01-07:
I have Xiao BLE (NRF) boards with an integrated common cathode RGB light. I'd really like to be able to change the PWM values for each color based on behaviors like 

- when I change layers
- when caps lock is on
- when the battery is low
- when the bt is connected, etc.

Obviously, most RGB is done with WS2812 or similar with a data line signaling, but what about PWM LEDs?

idesignstuff, 2023-01-31:
PWM LEDS that are on every Xiao BLE board...

NevPalmer, 2024-12-27:
I second this. Having the ability to switch simple GPIO controlled LEDs would be useful to provide basic indicators for layers, num lock, caps lock, etc. Simple HIGH/LOW activation would be good. PWM control would be even better.
This could potentially be made more power efficient than using WS2812 or similar, which consume power even when unlit.
It was possible to achieve with my Kyria keyboard when wired and running QMK. Now I have converted to wireless running ZMK I'm somewhat disappointed the capability doesn't appear to be available.

caksoylar, 2024-12-27:
https://github.com/caksoylar/zmk-rgbled-widget might be a good reference, for how to implement such a feature if not use directly.

(As an aside, I didn't implement PWM for above because I am not sure of the power use implications of enabling the PWM-related timers.)

NevPalmer, 2024-12-27:
> https://github.com/caksoylar/zmk-rgbled-widget might be a good reference, for how to implement such a feature if not use directly.
> 
Thanks for the reference. I'll take a look but may still be outside my ability to adapt and implement what I want.
I would like simple ability to toggle individual LEDs. One each to indicate if a specific layer, numlock, etc is active. I will repost if I manage to achieve.

peterjc, 2025-08-13:
https://github.com/BlueDrink9/zmk-poor-mans-led-indicator can also be used for this (but only attempts to control a single LED as might be found built into a controller). Support for HID indicator status like Caps Lock isn't (currently) supported though, layer changes etc are.