zmkfirmware/zmk #1624: Feature request: antecedent-morph (aka adaptive keys)

urob, 2023-01-10:
This is an idea for a new behavior, which I think could be really useful and open up all sorts of creative uses:

The `antecedent-morph` behavior would trigger an alternative binding if within `timeout-ms` the immediate predecessor matches the trigger condition. This would enable using adaptive keys where certain keys change their meaning if pressed immediately after another key, as popularized by the Hands Down layouts. 

For instance, in the Hands Down layout, `,A` send `UA`. Using the proposed behavior, this could be implemented by replacing `&kp A` with a `antecedent-morph` configured as follows: 
```
adaptive_A: adaptive_A {
    compatible=  "zmk,behavior-antecedent-morph";
    bindings = <&kp A>, <&morphed_A>;
    timeout-ms = <200>;
    antecedent = <&kp DOT>:
}
```
and where `&morphed_A` is a macro executing `BACKSPACE`, `BACKSPACE`, `U`, `A`.

caksoylar, 2023-12-11:
Would be fixed by #2042.

urob, 2025-01-10:
I ended up writing my own implementation as a module (https://github.com/urob/zmk-adaptive-key/), which works perfect for me. Leaving the issue open since there is a related open PR. Though if any of the maintainers wants to close it that's fine for me.