zmkfirmware/zmk #1633: BLE split timing synchronization

bendlas, 2023-01-24:
There seems to be a race condition for sending keys keys to the host, between the two halves of my corne:

#### Description

When pressing two keys on different halves close to the same time (or even farther apart during BLE hiccups), the keys often arrive out of order, which makes it pretty much impossible to type fast in BLE-split mode.

When connecting the halves with cable, the problem completely goes away, even when the host connection remains BT.

Apparently the `CENTRAL` half has a shorter delay, so e.g., pressing a modifier on my left half for a key on the right half works, but pressing a modifier on the right half for a key on the left half, usually fails.

#### Theorizing; questions

Being new to ZMK, I'd assume that the cause for this behavior is missing synchronization between the halves. When going over wire, it's not noticable, but the unpredictability of BLE exposes it.

Could this be fixed by the halves just waiting for each other, when detecting a keypress? Something like:

- Central detects (or receives) keypress
- Central waits on all (other) stations for timestamp >= press before assigning and sending a scancode

Is this a realistic szenario, or am I misunderstanding something (e.g. like each station sends its own scancodes via BT)?

Should such synchronization support be done only for BLE split, or could there be a scenario where wire protocol would profit from this as well? cc https://github.com/zmkfirmware/zmk/issues/58

petejohanson, 2023-01-24:
Plugging Ina TRRS cable does *nothing* from a communication/protocol perspective with ZMK today. When connected with the wire, are the two halves closer together? Anything removed between them?

The usual fix for this is bumping the radio TX power so the peripheral side successfully sends the key press events without re-transmits, etc. Have you tried that yet?

bendlas, 2023-01-24:
Interesting! That must have been a side-effect of the halves being closer together then.

Indeed, the Problem seems to go away when putting the halves within cable distance. Sorry for not checking that variable first.

> The usual fix for this is bumping the radio TX power so the peripheral side successfully sends the key press events without re-transmits, etc. Have you tried that yet?

No, I'll check it out, thanks!

bendlas, 2023-01-24:
Since my issue is not specific to BLE, I'll close this issue to reduce the noise.

Do you think it might still be worth pursuing better synchronisation, as outlined in the description @petejohanson ?

closed 2023-01-24 by bendlas