zmkfirmware/zmk #1637: Peripheral connects to wrong central

urob, 2023-01-27:
I noticed that when having multiple split boards, then the peripherals sometimes get confused and connect to the wrong central. Steps to reproduce:

- Board B: both sides sleeping
- Board A: reset peripheral A regardless of central A's state **or** awake peripheral A while central A is asleep **or** awake central A and peripheral A around the same time

In these situations, peripheral A will "connect" to central B, resulting in its key presses going unregistered.

---

## Log files

Note:
- Central A = `F8:3A:01:53:13:ED`
- Central B =`CB:21:AF:11:2B:F8`
- Peripheral A =`E4:C6:AA:AE:2C:27`

Here's the relevant log from Peripheral A. Note it connecting to `CB:21:AF:11:2B:F8` (Central B).
```
[00:00:00.797,698] <dbg> zmk: zmk_usb_get_conn_state: state: 1
[00:00:00.951,904] <dbg> zmk: zmk_usb_get_conn_state: state: 3
[00:00:06.094,329] <dbg> zmk: le_param_updated: CB:21:AF:11:2B:F8 (random): interval 12 latency 30 timeout 400
[00:00:09.140,136] <dbg> zmk: kscan_matrix_read: Sending event at 1,1 state on
[00:00:09.140,228] <dbg> zmk: zmk_kscan_process_msgq: Row: 1, col: 1, position: 16, pressed: true
[00:00:09.140,258] <dbg> zmk: split_listener:
[00:00:09.140,350] <dbg> zmk: send_position_state_callback: Error notifying -128
[00:00:09.236,145] <dbg> zmk: kscan_matrix_read: Sending event at 1,1 state off
[00:00:09.236,267] <dbg> zmk: zmk_kscan_process_msgq: Row: 1, col: 1, position: 16, pressed: false
```

Here's the relevant log from Central A. Note it not connecting to any peripheral.

```
[00:13:31.984,619] <dbg> zmk: split_central_disconnected: Disconnected: E4:C6:AA:AE:2C:27 (random) (reason 8)
[00:13:31.984,649] <dbg> zmk: release_peripheral_slot: Releasing peripheral slot at 0
[00:13:31.985,229] <dbg> zmk: start_scan: Scanning successfully started
[00:13:31.985,321] <dbg> zmk: disconnected: Disconnected from E4:C6:AA:AE:2C:27 (random) (reason 0x08)
[00:13:31.985,351] <dbg> zmk: disconnected: SKIPPING FOR ROLE 0
[00:13:32.001,037] <dbg> zmk: split_central_device_found: [DEVICE]: 3E:84:3C:90:29:E4 (random), AD evt type 3, AD data len 15, RSSI -85
[00:13:32.114,532] <dbg> zmk: split_central_device_found: [DEVICE]: 64:7F:7B:B9:D6:69 (random), AD evt type 0, AD data len 17, RSSI -79
[00:13:32.114,562] <dbg> zmk: split_central_eir_found: [AD]: 1 data_len 1
[00:13:32.114,562] <dbg> zmk: split_central_eir_found: [AD]: 10 data_len 1
[00:13:32.114,562] <dbg> zmk: split_central_eir_found: [AD]: 255 data_len 9
[00:13:32.116,851] <dbg> zmk: split_central_device_found: [DEVICE]: 0B:8A:AB:85:60:F5 (random), AD evt type 3, AD data len 31, RSSI -57
[00:13:32.493,988] <dbg> zmk: split_central_device_found: [DEVICE]: 76:70:4B:0E:43:C5 (random), AD evt type 0, AD data len 18, RSSI -84
[00:13:32.493,988] <dbg> zmk: split_central_eir_found: [AD]: 1 data_len 1
[00:13:32.494,018] <dbg> zmk: split_central_eir_found: [AD]: 10 data_len 1
[00:13:32.494,018] <dbg> zmk: split_central_eir_found: [AD]: 255 data_len 10
[00:13:32.596,984] <dbg> zmk: split_central_device_found: [DEVICE]: 41:19:B8:69:77:8B (random), AD evt type 0, AD data len 18, RSSI -85
[00:13:32.597,015] <dbg> zmk: split_central_eir_found: [AD]: 1 data_len 1
[00:13:32.597,015] <dbg> zmk: split_central_eir_found: [AD]: 10 data_len 1
[00:13:32.597,015] <dbg> zmk: split_central_eir_found: [AD]: 255 data_len 10
[00:13:34.223,144] <dbg> zmk: split_central_device_found: [DEVICE]: DC:22:9C:95:8D:6D (random), AD evt type 3, AD data len 8, RSSI -37
[00:13:34.349,121] <dbg> zmk: split_central_device_found: [DEVICE]: F6:79:9C:C7:F9:21 (random), AD evt type 3, AD data len 8, RSSI -85
[00:13:34.401,885] <dbg> zmk: split_central_device_found: [DEVICE]: 7A:DF:3F:51:0C:C5 (random), AD evt type 0, AD data len 17, RSSI -86
[00:13:34.401,885] <dbg> zmk: split_central_eir_found: [AD]: 1 data_len 1
[00:13:34.401,885] <dbg> zmk: split_central_eir_found: [AD]: 10 data_len 1
[00:13:34.401,916] <dbg> zmk: split_central_eir_found: [AD]: 255 data_len 9
[00:13:34.464,782] <dbg> zmk: split_central_device_found: [DEVICE]: FB:61:4B:F8:27:EC (random), AD evt type 3, AD data len 8, RSSI -84
[00:13:35.432,495] <dbg> zmk: split_central_device_found: [DEVICE]: 45:7D:A2:24:B9:3D (random), AD evt type 2, AD data len 31, RSSI -86
[00:13:35.612,518] <dbg> zmk: split_central_device_found: [DEVICE]: 49:F1:86:BE:05:83 (random), AD evt type 0, AD data len 18, RSSI -84
[00:13:35.612,548] <dbg> zmk: split_central_eir_found: [AD]: 1 data_len 1
[00:13:35.612,548] <dbg> zmk: split_central_eir_found: [AD]: 3 data_len 2
[00:13:35.612,579] <dbg> zmk: split_central_eir_found: [AD]: 255 data_len 9
[00:13:35.889,831] <dbg> zmk: split_central_device_found: [DEVICE]: D4:DA:F2:D9:4C:9D (random), AD evt type 3, AD data len 8, RSSI -84
[00:13:40.948,425] <dbg> zmk: split_central_device_found: [DEVICE]: 10:97:BD:7F:E4:C2 (public), AD evt type 0, AD data len 19, RSSI -88
[00:13:40.948,455] <dbg> zmk: split_central_eir_found: [AD]: 1 data_len 1
[00:13:40.948,455] <dbg> zmk: split_central_eir_found: [AD]: 255 data_len 14
[00:13:44.168,914] <dbg> zmk: split_central_device_found: [DEVICE]: C8:B9:9C:57:8D:EB (random), AD evt type 3, AD data len 8, RSSI -85
[00:13:50.903,900] <dbg> zmk: split_central_device_found: [DEVICE]: E4:C6:AA:AE:2C:27 (random), AD evt type 0, AD data len 25, RSSI -64
[00:13:50.903,930] <dbg> zmk: split_central_eir_found: [AD]: 1 data_len 1
[00:13:50.903,930] <dbg> zmk: split_central_eir_found: [AD]: 2 data_len 2
[00:13:50.903,930] <dbg> zmk: split_central_eir_found: [AD]: 7 data_len 16
```

@petejohanson mentioned plans for direct advertising as a way to fix this. Putting this up to gather the relevant logs.

xudongzheng, 2023-01-27:
For split keyboards, the central is responsible for initiating connections. It would be central B that initiates the connection to peripheral A and not vice versa.

While central B remembers a bond with peripheral B, it doesn't actually check the MAC address when scanning. As a result, it can end up connecting to another ZMK peripheral (peripheral A in this case).

This is issue #609 and is indirectly addressed in #836.

petejohanson, 2023-01-27:
Having the peripheral do directed advertising to only the address of the central that it previously connected to, should be a bulletproof way to be sure that only the paired central discovers that peripheral. 

That's an enhancement that has been on my list to help be sure centrals don't accidentally discover other advertising peripherals.

xudongzheng, 2023-01-27:
Ideally directed advertising should be a configurable option (perhaps enabled by default for `CONFIG_BT_MAX_PAIRED=1`).

One potential use case is having a split keyboard bond with multiple dongles (for instance, one at home and one at work) with `CONFIG_BT_MAX_PAIRED=2`. In such a scenario, the peripheral wouldn't know what device it should direct the advertisement to.

caksoylar, 2024-02-27:
I think this should be fixed after #1913, closing.

closed 2024-02-27 by caksoylar