zmkfirmware/zmk #1661: Sticky Key followed by Hold Tap does not activate Sticky Key

td0m, 2023-02-12:
For example: tapping  `&sk LSHIFT` followed by `&ht LCTRL N` does not uppercase the `n`.
I am not sure if this is a bug, or something related to my configuration?

My relevant config looks like this:
```
&sk {
    release-after-ms = <1000>;
};

/ {
  behaviors {
      ht: hold_tap {
            label = "hold_tap";
            compatible = "zmk,behavior-hold-tap";
            #binding-cells = <2>;
            flavor = "tap-preferred";
            tapping-term-ms = <210>;
            quick-tap-ms = <150>;
            global-quick-tap;
            bindings = <&kp>, <&kp>;
        };
  };
```

FriederRodewald, 2024-02-10:
I have the same issue - any solution yet?

FriederRodewald, 2024-02-13:
After investing some unnecessary hours into this issue, I think I finally found the issue:

caksoylar, 2024-02-13:
I don't think OS level sticky keys would have relation to the sticky key behavior in ZMK. Maybe it is related to the in general [wonky handling of key events](https://github.com/zmkfirmware/zmk/issues/759) that Gnome on Wayland before <45 has.

FriederRodewald, 2024-02-14:
Wow, thank you so much - the (temporary) solution for me was to `pkill ibus` as suggested in #759.

closed 2024-02-14 by td0m