zmkfirmware/zmk #1671: Random key stuck and repeat input one key value until stop responding

jackytrabbit, 2023-02-20:
Not sure if it's a ZMK issue but just found it on one of my recent builds.
Most of the time, it works great. But one issue happens maybe like twice in a day during regular typing. It suddently stucks in one key that I pressed and keeps generating key value, and then it stops responding. Removing the bluetooth device on PC and pressing 'BT_CLR' (or press 'reset' on keyboard'), it goes back to normal.  The key stuched every time is random, not just on one key.

jackytrabbit, 2023-02-21:
> 

> I'd be curious to see the logs when this occurs to better understand. Also useful to see your keymap; are there hold-taps on the keys that have issues, etc?

No hold-taps is enabled. Just basic keys, no macro, combo. I will enable usb logging and check the log file.

jackytrabbit, 2023-02-22:
It seems that the issue is gone after I reflashed firmware with usb logging enabled. I will close this issue. 
Thanks again for the great work! @petejohanson

closed 2023-02-22 by jackytrabbit

petejohanson, 2023-02-22:
Do beware of the extra power draw that leaving logging on incurs. I would definitely disable it again and confirm the issue is still resolved.

jackytrabbit, 2023-02-22:
> Do beware of the extra power draw that leaving logging on incurs. I would definitely disable it again and confirm the issue is still resolved.

Got it! Thanks for the heads-up.