zmkfirmware/zmk #1678: Question: How i can swap the key 1 => SHIFT+1

approached, 2023-02-22:
Hi @ all,

How i can swap the key `1` => `SHIFT+1`?

caksoylar, 2023-02-22:
Can you elaborate? If you just want to send `shift+1` on a single key press you can use `&kp EXCLAMATION` or equivalently `&kp LS(N1)` using [modifier functions](https://zmk.dev/docs/codes/modifiers#modifier-functions).

caksoylar, 2023-02-23:
You should be able achieve this with a [mod-morph](https://zmk.dev/docs/behaviors/mod-morph) with `bindings = <&kp EXCL>, <&kp N1>;` and `mods = <(MOD_LSFT|MOD_RSFT)>;`. I am afraid I am not familiar with Kinesis's editor to say if you can use it for this, or you'd need to edit the keymap file directly.

hyx0329, 2023-02-25:
example code

```
/ {
	behaviors {

		// multi-function media keys

		c_prev_rw: previous_rewind {
			compatible = "zmk,behavior-mod-morph";
			label = "PREVIOUS_REWIND";
			#binding-cells = <0>;
			bindings = <&kp C_PREVIOUS>, <&kp C_REWIND>;
			mods = <(MOD_LALT|MOD_RALT)>;
			keep-mods = <0>; // do not keep modifiers
		};

		c_next_ff: next_fastfoward {
			compatible = "zmk,behavior-mod-morph";
			label = "NEXT_FASTFOWARD";
			#binding-cells = <0>;
			bindings = <&kp C_NEXT>, <&kp C_FAST_FORWARD>;
			mods = <(MOD_LALT|MOD_RALT)>;
			keep-mods = <0>; // do not keep modifiers
		};

		// various keys for programmer dvorak

		dllr_tilde: dollar_tilde {
			compatible = "zmk,behavior-mod-morph";
			label = "DOLLAR_TILDE";
			#binding-cells = <0>;
			bindings = <&kp DOLLAR>, <&kp TILDE>;
			mods = <(MOD_LSFT|MOD_RSFT)>;
			keep-mods = <0>; // do not keep modifiers
		};
	};

	keymap0: keymap {
		compatible = "zmk,keymap";

		default_layer {
		...
		};
	};
};
```

approached, 2023-02-25:
Thank you for the comments. This works for me good:
```
#define MOD_MORPH(NAME, KEY, MOD_KEY) \
  mm_##NAME: mm_##NAME { \
    compatible = "zmk,behavior-mod-morph"; \
    label = #NAME; \
    #binding-cells = <0>; \
    bindings = <&kp KEY>, <&kp MOD_KEY>; \
    mods = <(MOD_LSFT|MOD_RSFT)>; \
  };

/ {
  behaviors {
    #include "macros.dtsi"

    // various keys for programmers
    MOD_MORPH(n1, EXCL, N1)
    MOD_MORPH(n2, AT_SIGN, N2)
    MOD_MORPH(n3, HASH, N3)
    MOD_MORPH(n4, DOLLAR, N4)
    MOD_MORPH(n5, PERCENT, N5)
    MOD_MORPH(n6, CARET, N6)
    MOD_MORPH(n7, AMPERSAND, N7)
    MOD_MORPH(n8, ASTERISK, N8)
    MOD_MORPH(n9, LEFT_PARENTHESIS, N9)
    MOD_MORPH(n0, RIGHT_PARENTHESIS, N0)
```

closed 2023-02-25 by approached