zmkfirmware/zmk #1685: Can't get SHIFT-SPACE to issue TAB

yanshay, 2023-02-25:
I am trying to get SPACE function as TAB when pressing SHIFT-SPACE. 
I am using mod morph as follows:
```
    spc_tab: space_tab {
            compatible = "zmk,behavior-mod-morph";
            label = "SPACE-TAB";
            #binding-cells = <0>;
            bindings = <&kp SPACE>, <&kp TAB>;
            mods = <(MOD_LSFT)>;
        };
```
However it issues a SHIFT-TAB and not a TAB.
This is what I see in karabiner event viewe:
```
  {
    "type": "down",
    "name": {"key_code":"left_shift"},
    "usagePage": "7 (0x0007)",
    "usage": "225 (0x00e1)",
    "misc": "flags left_shift"
  },
  {
    "type": "down",
    "name": {"key_code":"tab"},
    "usagePage": "7 (0x0007)",
    "usage": "43 (0x002b)",
    "misc": "flags left_shift"  <--
  },
  {
    "type": "up",
    "name": {"key_code":"left_shift"},
    "usagePage": "7 (0x0007)",
    "usage": "225 (0x00e1)",
    "misc": ""
  },
  {
    "type": "up",
    "name": {"key_code":"tab"},
    "usagePage": "7 (0x0007)",
    "usage": "43 (0x002b)",
    "misc": ""
  },
```
I even tried adding  `keep-mods = <0>;` and it didnt change a thing. 
so why is the mod being sent together with the key? it practically means I cant get to issue TAB using SHIFT-SPACE.

caksoylar, 2023-02-26:
This came up a couple of times in the Discord server, but only seems to be related to KA360 Pro somehow, e.g. https://discord.com/channels/719497620560543766/1055137812481060884/1055137812481060884. There is a corresponding issue at https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/83. One thing I might try since those issues came up is [disabling NKRO](https://github.com/yanshay/Adv360-Pro-ZMK/blob/V2.0/config/boards/arm/adv360/adv360_left_defconfig#L53) since that is one difference I noticed between usual ZMK setups and KA360.

yanshay, 2023-02-26:
I tried disabling NKRO and it didn't help.

To make sure I disabled properly, I am building locally using podman and changed that flag from 'y' to 'n' on both left and right keyboards in the local files.

Any other ideas?

ReFil, 2023-02-26:
I maintain the adv360 pro firmware, no idea why this is happening as all the keys and macro stuff is straight from ZMK main, there's only changes to the RGB code to add a couple unique effects, and minor tweaks to split connection handling. 

There's currently an open PR #1454 to add support for the ADV360 Pro in zmk main. @yanshay maybe try adding your macro to that firmware and trying again to see if it's the changes made to the zmk branch or a problem in the configuration files?

yanshay, 2023-02-26:
> There's currently an open PR #1454 to add support for the ADV360 Pro in zmk main. @yanshay maybe try adding your macro to that firmware and trying again to see if it's the changes made to the zmk branch or a problem in the configuration files?

Happy to try, how do I do that?

yanshay, 2023-02-27:
> There's currently an open PR #1454 to add support for the ADV360 Pro in zmk main. @yanshay maybe try adding your macro to that firmware and trying again to see if it's the changes made to the zmk branch or a problem in the configuration files?

@ReFil Happy to try, how do I do that? 
I only know how to use the Advantage-ZMK-Pro repo which includes only configuration. 
I tried pointing to the branch in the west.yaml (and I'm building locally using podman) and got an error  `devicetree error: /app/config/adv360.keymap:111 (column 47): parse error: expected number or parenthesized expression`, especially since my config file doesn't have so many lines, so I'm probably doing something wrong. 

Can you guide me how to build a firmware based on that branch with my config?

ReFil, 2023-02-27:
Apologies, i thought i typed out a guide but it must have not posted or something.

It's easiest if you use the VSCode and Docker method detailed on the ZMK docs, you can follow this guide: https://zmk.dev/docs/development/setup

Once you get to "Get source code" Use this link instead: https://github.com/ReFil/zmk.git
and then type `git checkout advantage360`

Once this is complete you want to add your macro that doesnt work into the keymap for the new board definition. you want to go to the directory you cloned in the "Get source code step" go to *zmk folder*/app/boards/arm/adv360pro/adv360pro.keymap

Do not just copy the keymap file from your config repo, the custom branch has some keys that are not compatible, that is why your compilation with podman didn't work.

yanshay, 2023-02-27:
@ReFil Thanks, I'll try that.
In the mean time I did some more research and tried using urob https://github.com/urob/adv360-demo-config.
What's strange is that while in Karabiner Event Viewer the events are the same, the actual behavior of the Tab using Shift-Space there is correct (so the way Tab should behave, for example in the Terminal) while with the standard Advantage Pro it is not (behaves differently, different to explain exactly what it does).
Are there elements of the keystrokes that are not shown in Karabiner Event Viewer?

yanshay, 2023-02-27:
> Apologies, i thought i typed out a guide but it must have not posted or something.
>
> It's easiest if you use the VSCode and Docker method detailed on the ZMK docs, you can follow this guide: [...]

@ReFil How do I build? The instructions require knowing board, pcb, left vs right, etc.

yanshay, 2023-02-27:
@ReFil Managed to build, I hope I did it correctly, the symptoms are the same. Shift + Mod-Morph on first press cause a shift to be sent (bug) and the second press isn't shifted (functions correctly).
With Urbo repo this doesn't happen.

Does it mean it's a bug with zmk and on all keyboards? Or something to do with the Advantage Pro 360 zmk configurations?

ReFil, 2023-02-28:
If it doesn't happen on the urob repo it sounds like there's a problem with the config files. I will continue to investigate

yanshay, 2023-03-03:
I made some progress here. 
It seems like the issue has to do with Karabiner Elements and not with urob repo vs. other zmk.
When in karabiner elements in the devices page I have "Modify events" enabled then the faulty behavior takes place. Without it enabled it works as expected.
For some reason when I switched to urob generated firmware (which caused a need to reset bluetooth settings) that toggle got disabled.
However, this doesn't solve my problem since once I disable it karabiner stops processing keyboard events and I loose configurations.
Note that this happens also when I clean all my configurations in karabiner elements so it isn't supposed to do anything.
 Does anyone know what does karabiner do that could cause that behavior?

yanshay, 2023-03-03:
But going back to the original event sequence, is the set of events the mod-morph generates is correct? 
The way I read the events sequence it is indeed shifted tab and not tab.
I'v seen all kinds of discussions about mod-morph ability to swallow mods (#1114) and masked-mod functionality that I see in the code but was removes from the attributes yaml. But couldn't find a solution to this issue.

divsmith, 2023-06-06:
I'm encountering this as well on a nice!nano in MacOS, both over usb and bluetooth. It seems as though ZMK isn't swallowing the held modifiers for the initial press of a mod-morphed key.

I have a mod-morph configured to send comma `,` normally and semicolon `;` when left shift is held:
```
comma_colon: COMMA_SEMICOLON {
        compatible = "zmk,behavior-mod-morph";
        label = "COMMA_SEMICOLON";
        #binding-cells = <0>;
        bindings = <&kp COMMA>, <&kp SEMI>;
        mods = <(MOD_LSFT)>;
};
```

Holding down left shift and pressing comma twice results in the following logs:
```
[
  {
    "type": "down",
    "name": {"key_code":"left_shift"},
    "usagePage": "7 (0x0007)",
    "usage": "225 (0x00e1)",
    "misc": "flags left_shift"
  },
  {
    "type": "down",
    "name": {"key_code":"semicolon"},
    "usagePage": "7 (0x0007)",
    "usage": "51 (0x0033)",
    "misc": "flags left_shift"
  },
  {
    "type": "up",
    "name": {"key_code":"left_shift"},
    "usagePage": "7 (0x0007)",
    "usage": "225 (0x00e1)",
    "misc": ""
  },
  {
    "type": "up",
    "name": {"key_code":"semicolon"},
    "usagePage": "7 (0x0007)",
    "usage": "51 (0x0033)",
    "misc": ""
  },
  {
    "type": "down",
    "name": {"key_code":"semicolon"},
    "usagePage": "7 (0x0007)",
    "usage": "51 (0x0033)",
    "misc": ""
  },
  {
    "type": "up",
    "name": {"key_code":"semicolon"},
    "usagePage": "7 (0x0007)",
    "usage": "51 (0x0033)",
    "misc": ""
  }
]
```
The first semicolon down is sent before left_shift up, resulting in `:;` instead of `;;`

fabianpage, 2023-09-15:
I get the same problem. (I have a combo which sends 1 without shift and F1 with shift.) 
When I press shift, the first F1 will be sent with shift, the following ones without.

mskvsk, 2023-11-03:
same thing

```
        right_br: right_and_brightness_inc {
                        compatible = "zmk,behavior-mod-morph";
                        label = "BRIGHTNESS_UP_DOWN";
                        #binding-cells = <0>;
                        bindings = <&kp RIGHT>, <&kp C_BRI_UP>;
                                mods = <MOD_LALT>;
                                keep-mods = <0>;
                };
```

When pressed with Alt, it sends Alt+C_BRI_UP.

I am running it on Corne-isn Zen.