zmkfirmware/zmk #1688: Question: Porting qmk keyboard config to zmk

asmodeus812, 2023-02-28:
Hi, i was wondering if there is a way, or even if it is supported, to port the following qmk board configuration to zmk https://github.com/qmk/qmk_firmware/pull/9944/. It is a bit convoluted since it uses multiplexing on the rows, but since i am not too familiar with zmk, would it be possible to implement it in a similar way as it was for qmk. I am willing to work on it, with some guidance in the right direction, thanks !

joelspadin, 2023-02-28:
It is possible to write new key scanning drivers to handle specialized matrices. Unfortunately, you won't be able to reference the QMK code for it when porting to ZMK due to licensing.

nmunnich, 2025-08-14:
Closing as completed, as it seems Joel answered your question to your satisfaction since there has been no further activity.

closed 2025-08-14 by nmunnich