zmkfirmware/zmk #1691: Tap dance with layer tap: Slow activation

infused-kim, 2023-03-01:
I noticed that a layer tap within a tap dance takes much longer to activate than having the layer tap without the tap dance.

Here is an example config:
```
  td_num_layer: td_num_layer {
            compatible = "zmk,behavior-tap-dance";
            label = "TAP_DANCE_NUM_LAYER";
            #binding-cells = <0>;
            tapping-term-ms = <250>;
            bindings = <&lt NUM DEL>, <&num_word>;
        };
```

## What I am expecting

- I quickly hold the `&td_num_layer` key and press `w` key to write `7`.
- The number appears immediately

## What is actually happening
- I quickly hold the `&td_num_layer` key and press `w` key to write `7`.
- The character  `w` appears instead
- The tap action of `&td_num_layer ` is executed

To make it work, I have to:
- Hold the `&td_num_layer` key
- Wait for brief moment 
- Press the `w` key
- The number `7` appears

## Log
Based on the log it looks like the layer tap behavior starts its timing only after the tap dance has been decided. But that's probably not what most users want.

```
[00:20:47.442,413] <dbg> zmk: split_central_notify_func: [NOTIFICATION] data 0x2001299b length 16
[00:20:47.442,443] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.442,443] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.442,474] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.442,474] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.442,474] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.442,504] <dbg> zmk: split_central_notify_func: data: 2
[00:20:47.442,504] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.442,504] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.442,535] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.442,535] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.442,565] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.442,565] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.442,565] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.442,596] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.442,596] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.442,596] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.442,687] <dbg> zmk: peripheral_event_work_callback: Trigger key position state change for 41
[00:20:47.442,718] <dbg> zmk: position_state_changed_listener: 41 bubble (no undecided hold_tap active)
[00:20:47.442,779] <dbg> zmk: zmk_keymap_apply_position_state: layer: 0 position: 41, binding name: TAP_DANCE_NUM_LAYER
[00:20:47.442,901] <dbg> zmk: on_tap_dance_binding_pressed: 41 created new tap dance
[00:20:47.442,901] <dbg> zmk: on_tap_dance_binding_pressed: 41 tap dance pressed
[00:20:47.442,932] <dbg> zmk: reset_timer: Successfully reset timer at position 41
[00:20:47.512,145] <dbg> zmk: kscan_matrix_read: Sending event at 0,2 state on
[00:20:47.512,237] <dbg> zmk: zmk_kscan_process_msgq: Row: 0, col: 2, position: 2, pressed: true
[00:20:47.512,268] <dbg> zmk: position_state_changed_listener: 2 bubble (no undecided hold_tap active)
[00:20:47.512,298] <dbg> zmk: position_state_down: combo: capturing position event 2
[00:20:47.512,298] <dbg> zmk: zmk_event_manager_handle_from: Listener captured the event
[00:20:47.547,332] <dbg> zmk: release_pressed_keys: combo: releasing position event 2
[00:20:47.547,363] <dbg> zmk: tap_dance_position_state_changed_listener: Tap dance interrupted, activating tap-dance at 41
[00:20:47.547,393] <dbg> zmk: on_hold_tap_binding_pressed: 41 new undecided hold_tap
[00:20:47.547,485] <dbg> zmk: zmk_keymap_apply_position_state: layer: 0 position: 2, binding name: KEY_PRESS
[00:20:47.547,515] <dbg> zmk: on_keymap_binding_pressed: position 2 keycode 0x7001A
[00:20:47.547,546] <dbg> zmk: hid_listener_keycode_pressed: usage_page 0x07 keycode 0x1A implicit_mods 0x00 explicit_mods 0x00
[00:20:47.547,576] <dbg> zmk: zmk_hid_implicit_modifiers_press: Modifiers set to 0x00
[00:20:47.547,576] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[00:20:47.642,181] <dbg> zmk: kscan_matrix_read: Sending event at 0,2 state off
[00:20:47.642,333] <dbg> zmk: zmk_kscan_process_msgq: Row: 0, col: 2, position: 2, pressed: false
[00:20:47.642,364] <dbg> zmk: position_state_changed_listener: 41 bubbling 2 up event
[00:20:47.642,395] <dbg> zmk: tap_dance_position_state_changed_listener: Ignore upstroke at position 2.
[00:20:47.642,456] <dbg> zmk: zmk_keymap_apply_position_state: layer: 0 position: 2, binding name: KEY_PRESS
[00:20:47.642,517] <dbg> zmk: on_keymap_binding_released: position 2 keycode 0x7001A
[00:20:47.642,547] <dbg> zmk: hid_listener_keycode_released: usage_page 0x07 keycode 0x1A implicit_mods 0x00 explicit_mods 0x00
[00:20:47.642,547] <dbg> zmk: zmk_hid_implicit_modifiers_release: Modifiers set to 0x00
[00:20:47.642,578] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[00:20:47.667,419] <dbg> zmk: split_central_notify_func: [NOTIFICATION] data 0x2001299b length 16
[00:20:47.667,419] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.667,419] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.667,449] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.667,449] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.667,480] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.667,480] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.667,480] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.667,510] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.667,510] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.667,510] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.667,541] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.667,541] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.667,541] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.667,572] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.667,572] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.667,572] <dbg> zmk: split_central_notify_func: data: 0
[00:20:47.667,663] <dbg> zmk: peripheral_event_work_callback: Trigger key position state change for 41
[00:20:47.667,694] <dbg> zmk: position_state_changed_listener: 41 bubble undecided hold-tap keyrelease event
[00:20:47.667,694] <dbg> zmk: tap_dance_position_state_changed_listener: Ignore upstroke at position 41.
[00:20:47.667,755] <dbg> zmk: zmk_keymap_apply_position_state: layer: 0 position: 41, binding name: TAP_DANCE_NUM_LAYER
[00:20:47.667,846] <dbg> zmk: on_tap_dance_binding_released: 41 tap dance keybind released
[00:20:47.667,907] <dbg> zmk: decide_hold_tap: 41 decided tap (tap-preferred decision moment key-up)
[00:20:47.667,938] <dbg> zmk: on_keymap_binding_pressed: position 41 keycode 0x7004C
[00:20:47.667,968] <dbg> zmk: hid_listener_keycode_pressed: usage_page 0x07 keycode 0x4C implicit_mods 0x00 explicit_mods 0x00
[00:20:47.667,968] <dbg> zmk: zmk_hid_implicit_modifiers_press: Modifiers set to 0x00
[00:20:47.668,029] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[00:20:47.668,121] <dbg> zmk: on_keymap_binding_released: position 41 keycode 0x7004C
[00:20:47.668,151] <dbg> zmk: hid_listener_keycode_released: usage_page 0x07 keycode 0x4C implicit_mods 0x00 explicit_mods 0x00
[00:20:47.668,151] <dbg> zmk: zmk_hid_implicit_modifiers_release: Modifiers set to 0x00
[00:20:47.668,182] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[00:20:47.668,640] <dbg> zmk: on_hold_tap_binding_released: 41 cleaning up hold-tap
[00:21:00.535,949] <dbg> zmk: vddh_sample_fetch: ADC raw 1206 ~ 4415 mV => 100%
[00:21:54.057,128] <dbg> zmk: kscan_matrix_read: Sending event at 3,3 state on
[00:21:54.057,250] <dbg> zmk: zmk_kscan_process_msgq: Row: 3, col: 3, position: 36, pressed: true
[00:21:54.057,312] <dbg> zmk: position_state_changed_listener: 36 bubble (no undecided hold_tap active)
[00:21:54.057,342] <dbg> zmk: position_state_down: combo: capturing position event 36
[00:21:54.057,373] <dbg> zmk: zmk_event_manager_handle_from: Listener captured the event
[00:21:54.092,437] <dbg> zmk: release_pressed_keys: combo: releasing position event 36
[00:21:54.092,498] <dbg> zmk: zmk_keymap_apply_position_state: layer: 0 position: 36, binding name: GRESC_GUI
[00:21:54.092,620] <dbg> zmk: on_hold_tap_binding_pressed: 36 new undecided hold_tap
[00:21:54.257,720] <dbg> zmk: decide_hold_tap: 36 decided hold-timer (tap-preferred decision moment timer)
[00:21:54.257,751] <dbg> zmk: on_keymap_binding_pressed: position 36 keycode 0x700E3
[00:21:54.257,781] <dbg> zmk: hid_listener_keycode_pressed: usage_page 0x07 keycode 0xE3 implicit_mods 0x00 explicit_mods 0x00
[00:21:54.257,812] <dbg> zmk: zmk_hid_register_mod: Modifier 3 count 1
[00:21:54.257,843] <dbg> zmk: zmk_hid_register_mod: Modifiers set to 0x08
[00:21:54.257,843] <dbg> zmk: zmk_hid_implicit_modifiers_press: Modifiers set to 0x08
[00:21:54.257,843] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
```

infused-kim, 2023-03-05:
I just found a workaround for this. By nesting the tap dance within the layer tap / hold-tap, the delay disappears.

Since this use-case is being used in the documentation, perhaps it would make sense to update the example with this structure that is working much better?

```
        /*
        * Tap dancing nav layer
        *
        * Usage: &td_nav_layer NAV 0
        * Tap: RET
        * Double Tap: NAV_WORD layer
        * Triple-Tap-Hold: RET repeated
        * Hold: NAV layer
        */
        td_nav_layer_inner: td_nav_layer_inner {
            compatible = "zmk,behavior-tap-dance";
            label = "td_nav_layer_inner";
            #binding-cells = <0>;
            tapping-term-ms = <250>;
            bindings = <&kp RET>, <&nav_word>, <&kp RET>;
        };
        td_nav_layer: td_nav_layer {
            compatible = "zmk,behavior-hold-tap";
            label = "td_nav_layer";
            #binding-cells = <2>;
            flavor = "tap-preferred";
            tapping-term-ms = <250>;
            quick-tap-ms = <225>;
            bindings = <&mo>, <&td_nav_layer_inner>;
        };
```

kurtis-lew, 2023-03-05:
I think it would be worthwhile for the testing team to check if nesting tap-dances inside hold-taps, rather than hold-taps inside tap-dances, is a better implementation of their combination in all use cases. 

Out of curiosity, could you provide debug logs so we can compare them with your previous setup to determine why making the hold-tap as the parent behavior improves the activation timing?

yanshay, 2023-03-12:
This is similar to #1672

yanshay, 2023-03-13:
I didn't check but I guess this is caused by the same reasons for issue as #1701 - order in CMakeList that prevents tap-dance from being interrupted. And there are other combinations that don't work with tap-dance, probably because of that.