zmkfirmware/zmk #1704: Devicetree error: parse error: expected number or parenthesized expression

MykolaDevPy, 2023-03-12:
Hello, I'm trying to build my keyboard and after some changes in .config file I cant anymare to compile my firmware, here my code: 
```
/*
 * Copyright (c) 2020 The ZMK Contributors
 *
 * SPDX-License-Identifier: MIT
 */

#include <behaviors.dtsi>
#include <dt-bindings/zmk/keys.h>
#include <dt-bindings/zmk/bt.h>

#define AGRV &uc 00E0 00C0 // à À
#define CCED &uc 00E7 00C7 // ç Ç
#define EGRV &uc 00E8 00C8 // è È
#define EAIG &uc 00E9 00C9 // é É
#define UGRV &uc 00F9 00D9 // ù Ù

/ {
        keymap {
                compatible = "zmk,keymap";

                default_layer {
// -----------------------------------------------------------------------------------------
// |  TAB |  A  |  Z  |  E  |  R  |  T  |   |  Y  |  U   |  I  |  O  |  P  | BKSP |
// | CTRL |  Q  |  S  |  D  |  F  |  G  |   |  H  |  J   |  K  |  L  |  ;  |  '   |
// | SHFT |  W  |  X  |  C  |  V  |  B  |   |  N  |  M   |  ,  |  .  |  /  | ESC  |
//                    | GUI | LWR | SPC |   | ENT | RSE  | ALT |
                        bindings = <
   &kp TAB   &kp A &kp Z &kp E &kp R &kp T   &kp Y &kp U  &kp I     &kp O   &kp P    &kp BSPC
   &kp LCTRL &kp Q &kp S &kp D &kp F &kp G   &kp H &kp J  &kp K     &kp L   &kp SEMI &kp SQT
   &kp LSHFT &kp W &kp X &kp C &kp V &kp B   &kp N &kp M  &kp COMMA &kp DOT &kp FSLH &kp ESC
                &kp LGUI &mo 1 &kp SPACE      &kp RET &mo 2 &kp RALT
                        >;
                };
                lower_layer {
// -----------------------------------------------------------------------------------------
// |  TAB |  à  |  è  |  é  |     |     |   |     |  ù  | ç   |     |      | BKSP  |
// | LCTRL| BT1 | BT2 | BT3 | BT4 | BT5 |   | LFT | DWN |  UP | RGT | F11  | F12   |
// | SHFT | F1  | F2  | F3  | F4  | F5  |   | F6  | F7  |  F8 | F9  |  F10 | BTCLR |
//                    | GUI |     | SPC |   | ENT |     | ALT |
                        bindings = <
   &kp TAB    &kp AGRV     &kp EGRV     &kp EAIG     &trans       &trans         &trans   &kp UGRV &kp CCED &trans    &trans  &kp BSPC
   &kp LCTRL  &bt BT_SEL 0 &bt BT_SEL 1 &bt BT_SEL 2 &bt BT_SEL 3 &bt BT_SEL 4   &kp LEFT &kp DOWN &kp UP   &kp RIGHT &kp F11 &kp F12
   &kp LSHFT  &kp F1       &kp F2       &kp F3       &kp F4       &kp F5         &kp F6   &kp F7   &kp F8   &kp F9    &kp F10 &bt BT_CLR
                          	        &kp LGUI     &trans       &kp SPACE           &kp RET  &trans   &kp RALT
                        >;
                };  // ->  It tells that there is an arror devicetree error: parse error: expected number or parenthesized expression
                raise_layer {
// -----------------------------------------------------------------------------------------
// |  TAB |  @  |  %  |  (  |  )  | "|" |   |  =  |  +  |  1  |  2  |  3  | BKSP |
// | CTRL |  #  |  $  |  [  |  ]  |  &  |   |  -  |  0  |  4  |  5  |  6  |  `   |
// | SHFT |     |  !  |  {  |  }  |  \  |   |  _  |  *  |  7  |  8  |  9  |  ~   |
//                    | GUI |     | SPC |   | ENT |     | ALT |
                        bindings = <
   &kp  TAB   &kp AT         &kp PRCN    &kp LPAR  &kp RPAR  &kp PIPE      &kp EQUAL  &kp PLUS    &kp N1 &kp N2 &kp N3 &kp BSPC
   &kp LCTRL  &kp HASH       &kp DOLLAR  &kp LBKT  &kp RBKT  &kp AMPS      &kp MINUS  &kp N0      &kp N4 &kp N5 &kp N6 &kp GRAVE
   &kp LSHFT  &trans         &kp EXCL    &kp LBRC  &kp RBRC  &kp BSLH      &kp UNDER  &kp ASTRKS  &kp N7 &kp N8 &kp N9 &kp TILDE
                    	                   &kp LGUI  &trans   &kp SPACE       &kp RET   &trans    &kp RALT
                        >;
                };
        };
};
```

in line 46 there is an error. I added an indication. Can you help me please?

caksoylar, 2023-03-12:
Hi, you are defining `#define AGRV &uc 00E0 00C0 // à À` and then using it like `&kp AGRV`, which is wrong in two ways:
- `&uc` is not a predefined behavior in ZMK, if you use a fork or a custom behavior we'd need to see that
- `&kp AGRV` expands to `&kp &uc 00E0 00C0` which doesn't work because you can't use other node labels like `&uc` as parameters to behaviors

In general it is easier to get and provide support about user questions at the [Discord server](https://zmk.dev/community/discord/invite) so you could consider posting there next time, or to follow up on this.

MykolaDevPy, 2023-03-12:
Ok thank you, I find my error! But now I have another problem here:
```
                raise_layer {
// -----------------------------------------------------------------------------------------
// |  TAB |  @  |  %  |  (  |  )  | "|" |   |  =  |  +  |  1  |  2  |  3  | BKSP |
// | CTRL |  #  |  $  |  [  |  ]  |  &  |   |  -  |  0  |  4  |  5  |  6  |  `   |
// | SHFT |     |  !  |  {  |  }  |  \  |   |  _  |  *  |  7  |  8  |  9  |  ~   |
//                    | GUI |     | SPC |   | ENT |     | ALT |
                        bindings = <
   &kp TAB    &kp AT    &kp PRCN    &kp LPAR  &kp RPAR  &kp PIPE      &kp EQUAL  &kp PLUS    &kp N1  &kp N2  &kp N3  &kp BSPC
   &kp LCTRL  &kp HASH  &kp DOLLAR  &kp LBKT  &kp RBKT  &kp AMPS      &kp MINUS  &kp N0      &kp N4  &kp N5  &kp N6  &kp GRAVE
   &kp LSHFT  &trans    &kp EXCL    &kp LBRC  &kp RBRC  &kp BSLH      &kp UNDER  &kp ASTRKS  &kp N7  &kp N8  &kp N9  &kp TILDE
                    	            &kp LGUI  &trans  &kp SPACE        &kp RET   &trans    &kp RALT
                        >;
                };
```

It indicate me that I have parser error after `&kp PLUS`

caksoylar, 2023-03-13:
In this case I am guessing it is due to `PRCN`, since that isn't a ZMK keycode: https://zmk.dev/docs/codes#symbols--punctuation

The column number you get from the error message corresponds to the column after preprocessor expands the keycodes, so it won't be accurate.

MykolaDevPy, 2023-03-13:
Thank you very much! Now it works

closed 2023-03-13 by MykolaDevPy