zmkfirmware/zmk #1715: Key Repetition Behavior

xudongzheng, 2023-03-18:
Generally the operating system is responsible for handling key repetition behavior. However, it may be desirable to define a behavior that repeats a key at a configurable rate. This can be particularly useful for games where multiple key presses are handled differently from a single key press that's held.

urob, 2023-03-18:
Would this work? https://github.com/zmkfirmware/zmk/pull/1414

jjangga0214, 2023-03-18:
Would macro be not sufficient for the purpose?
https://zmk.dev/docs/behaviors/macros

jjangga0214, 2023-03-18:
As a workaround for now, you may use [kmonad](https://github.com/kmonad/kmonad).
There is `cmd-button` that executes arbitrary command(e.g. any script written by any language you want) from OS level.
You can write your script to generate/deactivate repetitive virtual keypresses when kmonad gets a certain keypress from zmk.
For example, AFAIK, macOS provides such a feature to generate virtual keypresses from software by Swift SDK.

xudongzheng, 2023-03-18:
> Would this work? #1414

Thanks. Somehow that didn't come up while searching.

> As a workaround for now, you may use [kmonad](https://github.com/kmonad/kmonad).

Very interesting, will look into that.

closed 2023-03-18 by xudongzheng