zmkfirmware/zmk #1716: feat: can we know which input source(current language) from OS?

jjangga0214, 2023-03-18:


joelspadin, 2023-03-18:
I don't believe this is possible without writing a custom PC application that communicates this to the keyboard.

jjangga0214, 2023-03-19:
@joelspadin Thanks for the response. I feel good to hear that it's technically possible. If this feature is developed, it will benefit a lot of people using multiple languages.

I found a tiny, but very useful project [macism](https://github.com/laishulu/macism/issues).
It can get the current activated input source name on macOS.

```bash
➜  macism
org.youknowone.inputmethod.Gureum.system
➜  macism
com.apple.keylayout.UnicodeHexInput
➜  macism
com.apple.keylayout.ABC
➜  macism
com.apple.inputmethod.Korean.2SetKorean
```

I haven't found yet, but I believe there're similar projects for Linux and Windows.
Then we may run background deamon program (like those programs shown in the status bar) that lets ZMK know when input source changes. 

[image]

And I imagine we will be able to write our keymap file like this.

```
        layer_english {
            input_source = "com.apple.keylayout.ABC"
            bindings = <&kp A &kp B>;
        };

        layer_korean {
            input_source = "com.apple.inputmethod.Korean.2SetKorean"
            bindings = <&kp B &kp A>;
        };
```

The layer would automatically change by input source, which frees people from language-related input issues.

I wondew how you think.

Sincerely,

Halverson-Jason, 2023-06-14:
So macism when ran in a shell will display the current activated input source, but the problem is that ZMK doesn't have access to the shell layer of the OS. What I believe would need to happen is this:

1. Background worker on OS scanning to see if a zmk keyboard was plugged in. 
2. When background worker finds keyboard it sends a specified command to the keyboard
3. Based on the command sent the keyboard can then store this as a variable to be used

Step one is possible to build, as this is in itself isnt a lot of code it is very specific to our use case I don't believe this exists and would have to be built. 

Step two is the problem, as far as I know, ZMK has a unidirectional HID link to the OS and does not have any method to receive commands without a bidirectional link. I think this would include adding a new HID report for incoming data to ZMK,  and implementing the handler in ZMK. You might have to modify the HID descriptor in the OS to include the new report that was added to ZMK. 

Overall this is a complex ask and not sure if the maintainers would be willing to add this to the backlog of work based on the technical challenge of implementing.

Hope this helps you understand the size of the ask and the issues surrounding the task!

hxse, 2023-08-31:
This question similar https://github.com/KMKfw/kmk_firmware/issues/869
In qmk, i need `raw_hid_receive` to receive system signal.
In kmk i need `Serialace`  to receive system signal.
I hope to do similar things with zmk.