zmkfirmware/zmk #1717: feat: macro parameter

jjangga0214, 2023-03-18:
Hi!

I think it'd be convenient if macro supports parameter(s).

```
        m: macro_with_param {
            label = "macro_with_param";
            compatible = "zmk,behavior-macro";
            #binding-cells = <0>;
            bindings
                = <&macro_tap &do_something &kp>; /* Here goes parameter*/
        };

        layer_default {
            bindings = <&m A &m B>
        }; 
```

For some situations, this would remove repetitive macros significantly.

joelspadin, 2023-03-18:
See https://github.com/zmkfirmware/zmk/pull/1232

It's not possible to implement it exactly as you'd written above because it will fail to parse if `&kp` isn't followed by a parameter, but one way we could implement this is to have special macro behaviors that modify the following (non-macro) behavior and indicate to replace its parameters with the parameters given to the macro.

For example:
```
wt: wrapped_tap {
    compatible = "zmk,behavior-macro-two-param";
    #binding-cells = <2>;
    bindings = 
        <&macro_press &macro_param1 &kp 0>
        <&macro_tap &macro_param2 &kp 0>
        <&macro_release &macro_param1 &kp 0>;
};

...
default_layer {
    bindings = <&wt A B>;
    // evaluates to &macro_press &kp A &macro_tap &kp B &macro_release &kp A
};
```

We would need a way to allow passing the 1st or 2nd macro parameter to the 1st or 2nd binding behavior. We could use a set of fixed names like

- `&macro_param1_1st` replace 1st param with macro param 1
- `&macro_param1_2nd` replace 2nd param with macro param 1
- `&macro_param2_1st` replace 1st param with macro param 2
- `&macro_param2_2nd` replace 2nd param with macro param 2
- `&macro_param_both` equivalent to `&macro_param1_1st &macro_param2_2nd`
- `&macro_param1` alias for `&macro_param1_1st` (shortcut for bindings that only take one param)
- `&macro_param2` alias for `&macro_param2_1st`

Alternatively, we could have a single behavior with parameters to control the replacement, but that could be confusing.

theol0403, 2023-06-22:
This can be closed due to https://github.com/zmkfirmware/zmk/pull/1232 !

closed 2023-06-22 by petejohanson