zmkfirmware/zmk #1718: Modifiers are triggered too fast for behaviors for iPadOS / iOS

tyalie, 2023-03-19:
I make use of mod morph extensively in order to adapt my layout to work with IBM german keyboard layouts. e.g. the standard <kbd>.</kbd> key, which produces > when using shift isn't readily available on German so I use mod morph to trigger the right keys in order to reproduce that behavior. This works great on Linux & co, but I struggle with it on iPadOS.

## What should happen?
When I press <kbd>RIGHT SHIFT</kbd> + <kbd>.</kbd> it should print out >. This is archived by triggering <kbd>LEFT SHIFT</kbd> + <kbd></></kbd> (on German layout). 

## Actual behavior (on iPadOS)?
Everything works when I use <kbd>LEFT SHIFT</kbd> + <kbd>.</kbd>, but it falls apart when I used <kbd>RIGHT SHIFT</kbd> + <kbd>.</kbd>. 

Initially the unshifted <kbd></></kbd> will be emitted (printing `<` instead of `>`) and only after a second trigger iPadOS will pick up on the modifier and continue.

## Ideas
It seems to me that the switch between <kbd>RIGHT SHIFT</kbd> and <kbd>LEFT SHIFT</kbd> which happens during triggering of mod morph happens far to fast and the keyboard input stack on iPadOS refuses to pick it up (maybe even by choice? who knows). Adding a small delay between <kbd>LEFT SHIFT</kbd> and the mod key (here <kbd></></kbd>) presumably will fix this behavior. 

I've not been able yet to test a solution as this seems to entail a few changes to the stack and the way buttons and modifiers are announced as this isn't necessarily a mod morph exclusive problem. Other behaviors which quickly change modifiers presumably run into similar issues.

tyalie, 2023-03-19:
Okay. I've been able to confirm my hypothesis. I've modified `on_mod_morph_binding_pressed` to the following:

```c
static int on_mod_morph_binding_pressed(struct zmk_behavior_binding *binding,
                                        struct zmk_behavior_binding_event event) {
    const struct device *dev = device_get_binding(binding->behavior_dev);
    const struct behavior_mod_morph_config *cfg = dev->config;
    struct behavior_mod_morph_data *data = dev->data;

    if (data->pressed_binding != NULL) {
        LOG_ERR("Can't press the same mod-morph twice");
        return -ENOTSUP;
    }

    if (zmk_hid_get_explicit_mods() & cfg->mods) {
        zmk_hid_masked_modifiers_set(cfg->masked_mods);
        zmk_endpoints_send_report(HID_USAGE_KEY);
        k_sleep(K_MSEC(1));
        data->pressed_binding = (struct zmk_behavior_binding *)&cfg->morph_binding;
    } else {
        data->pressed_binding = (struct zmk_behavior_binding *)&cfg->normal_binding;
    }
    return behavior_keymap_binding_pressed(data->pressed_binding, event);
}
```

Sending the keyboard report and waiting for an approx. 1ms delay resolve this issue. I just wanna state again, that this isn't the best solution as it is independent from mod morph, but it helps for now.

proostas, 2024-03-03:
Encountered the same issue on iPadOS. No modifier mentioned in keep-mods gets picked up by OS. Any headway with this issue? Mod-morph is unusable with iPadOS.