zmkfirmware/zmk #1719: Macro within a Tap Dance

Railang, 2023-03-20:
So I have a Macro that is known working, however when I put it into a Tapdance sequence it no longer works.
The end goal is when you tap c it enters "C" and when you hold C it enters CTRL + C.

macro_copy: macro_copy{
compatible = "zmk,behavior-macro";
label = "macro_copy";
#binding-cells = <0>;
bindings = <&macro_press &kp LCTRL>
, <&macro_press &kp C>
, <&macro_release &kp C>
, <&macro_release &kp LCTRL>;
};

td0: tap_dance_0 {
        compatible = "zmk,behavior-tap-dance";
        label = "TAP_DANCE_0";
        #binding-cells = <0>;
        tapping-term-ms = <100>;
        bindings = <&kp C>, <&macro_copy>;
    };

bindings = <
&td0

caksoylar, 2023-03-21:
> The end goal is when you tap c it enters "C" and when you hold C it enters CTRL + C.

You'd want to use a custom [hold-tap ](https://zmk.dev/docs/behaviors/hold-tap)for that, not tap dance. Tap dance will invoke the second behavior if you tap that key twice.

In this case you don't need to use a macro, you can use a keycode with [modifier functions](https://zmk.dev/docs/codes/modifiers#modifier-functions). So you could simply use `&mt LC(C) C` in your keymap.

caksoylar, 2023-03-21:
That's good, I'd recommend you use the [ZMK Discord](https://zmk.dev/community/discord/invite) for user questions where you can get faster help from a larger audience of users for the next time.

closed 2023-03-21 by caksoylar