zmkfirmware/zmk #1724: Empty repo was initialized when installing ZMK

marcus571, 2023-03-28:
I run ` bash -c "$(curl -fsSL https://zmk.dev/setup.sh)"` on windows git bash.

No matter which board and shield I choose, I get this:

`Downloading config file (https://raw.githubusercontent.com/zmkfirmware/zmk/main/app/boards/shields/contra/contra.conf)
Could not find it, falling back to https://raw.githubusercontent.com/zmkfirmware/zmk/main/app/boards/shields/contra/contra.conf
Downloading keymap file (https://raw.githubusercontent.com/zmkfirmware/zmk/main/app/boards/shields/contra/contra.keymap)
Could not find it, falling back to https://raw.githubusercontent.com/zmkfirmware/zmk/main/app/boards/shields/contra/contra.keymap
Warning: Could not find a keymap file to download!
~/zmk-test`
`
Initialized empty Git repository in C:/Users/Marcus/zmk-test/.git/
`

and an empty repo is initialized without .conf and .keymap files.

Did I do anything wrong?

joelspadin, 2023-03-28:
From the error, it looks like it was unable to download the files from GitHub. Have you tried running the script again to see if it was just a temporary network issue?

marcus571, 2023-03-29:
There is something wrong with my DNS settings. Problem solved, thank you.

closed 2023-03-29 by petejohanson