zmkfirmware/zmk #1732: Forward mouse events from USB over Bluetooth

MonsieurLanza, 2023-04-05:
Hello,

I may have a edge case here, so context:
- I am left handed and use a left handed ergonomic mouse -> (Contour Unimouse Wireless) which has a 2.4Ghz dongle.
- I often switch between 3 or more PCs.
- I use a bépo (sort of french dvorak) layout.

I currently use a USB ergodox. Every switch is a pain in the a**, so I ended up using barrier/input-leap, which is not a smooth experience (layout incompatibilities), and is not very conform on a professional laptop.

So the perfect combination would be a bluetooth (split) keyboard, with mouse dongle plugged in a keyboard’s USB port, and have ZMK forward mouse HID event over Bluetooth. A keyboard switch of bluetooth bond would also switch the mouse. 

I can also imagine, using a bluetooth mouse, being able to directly pair the mouse with the keyboard.

So questions and consideration :
- Is that feasible at all ? (I think so, but who knows)
- Dongle consumption may hurt.
- Where do I start ?

callym, 2023-04-05:
I've been looking for something that'd be able to do this for a while.

My use-case is that I have 3 devices I want to switch between but my bluetooth mouse only lets me switch between two, so being able to connect the mouse to a keyboard that can then switch between 3+ devices and forward the mouse connection would be great. 

I think it's technically possible to transparently forward the HID events, not sure what processing would need to be done on them to merge the mouse and keyboard events?

xudongzheng, 2023-04-25:
Connecting Bluetooth mouse would be the easiest and not require additional hardware. Stock ZMK central supports connecting to one ZMK peripheral, #836 extends that to multiple ZMK peripherals, and extending that to support generic HID peripherals is a matter of implementation. One does need to consider the total amount of CPU/radio time available. Connecting too many peripherals simultaneously will likely increase latency.

Integrating a USB HID device would require USB host support. That's not yet in Zephyr but appears to be under development with new USB stack zephyrproject-rtos/zephyr#42066. In terms of hardware, the easiest would be Nordic nRF52 with a chip like MAX3421E for USB host support. There are lots of alternatives such as using RP2040 for USB host https://github.com/sekigon-gonnoc/Pico-PIO-USB and defining your own communication protocol, or reimplementing the RP2040 PIO USB parts in Zephyr and using Zephyr's split host/controller setup for Bluetooth.

USB host support can be useful for other HID devices (foot pedals, joysticks), MIDI input, among other stuff. It can also be useful for a dongle similar to the Hasu USB to USB converter and allow ZMK to be used for arbitrary USB keyboards.

xudongzheng, 2025-10-22:
You can keep an eye on https://github.com/zephyrproject-rtos/zephyr/issues/95661 for USB host support in Zephyr. That would need to be in place for ZMK to support USB keyboard/mouse without introducing significant complexity.