zmkfirmware/zmk #1735: conditional macro

akira-toriyama, 2023-04-06:
Thanks for the great zmk.

I want to run if the condition is met on key release.
Used in combination with `&macro_pause_for_release`.

- `&time_pressed` 
- `&no_other_key_is_pressed`

```
conditional_macro: conditional_macro {
    label = "conditional_macro";
    compatible = "zmk,behavior-macro";
    #binding-cells = <0>;
    bindings
        = <&macro_pause_for_release>
        , <&macro_tap &time_pressed<1000ms &kp A>
        , <&macro_tap &no_other_key_is_pressed &kp A>;
};
```

or

```
bindings
    = <&macro_pause_for_release &time_pressed<1000>
    , <&macro_tap &kp A>;

bindings
    = <&macro_pause_for_release &no_other_key_is_pressed>
    , <&macro_tap &kp A>;
```

caksoylar, 2023-04-06:
Depending on your exact use case this kind of conditional branching can be achieved through the [hold-tap behavior](https://zmk.dev/docs/behaviors/hold-tap). For instance `tap-preferred` flavor decides hold vs. tap only conditioned on the duration of the press. `hold-preferred` triggers based on a combination of press duration and other keys getting pressed. `retro-tap` could also be useful.

akira-toriyama, 2023-04-07:
@caksoylar 

I think I can achieve this with `tap-hold`.
Thank you very much.

closed 2023-04-07 by akira-toriyama