zmkfirmware/zmk #1743: feature(split): add split-side encoder support

auipga, 2023-04-08:
Please add support for encoders on the peripheral side.

There is already a PR [here](https://github.com/zmkfirmware/zmk/pull/728)

Amit0617, 2023-04-20:
hello @auipga I am coming here from bountysource. Do you intend to get that feature merged by authors of repository or you just want to get that to be the part of the code? I can help for the latter case where I can fork the repo add the patch from the PR.

petejohanson, 2023-04-20:
Just FYI, I've been actively working on the necessary refactors before #728 can be merged:

The planned sequence is:

* #1659 (merged)
* #1758 (close, @joelspadin has been helping review and provide feedback)
* #1039 (I've rebased on top of #1758, but Joel had some good ideas for how to handle one design challenge, so potentially some more work needed)
* #728 last

The reason for the sequence is to be sure the sensor foundations are actually nicely designed to handle the split encoder work, which should end up being once smaller once everything is said and done.

maximus5684, 2023-04-24:
> The reason for the sequence is to be sure the sensor foundations are actually nicely designed to handle the split encoder work, which should end up being once smaller once everything is said and done.

Please let me know if I can help with this as I'm somewhat familiar with the ZMK codebase and have a SOFLE v2 with dual encoders.

autoferrit, 2023-05-09:
this is good to see progress. I tried to merge in the PR's in my own fork to get it working, but I think #1039 added enough changes to make it difficult for me to see how to merge #728 for my own uses. Is it possible to also rebase #728 on top of 1039 for those of us that are trying to keep up to date and have this feature enabled ourselves? 
I understand if that is complicated, I just thought I would ask as I couldn't really resolve the rebase myself.

petejohanson, 2023-06-21:
Ok, so I've created #1841 which is the rework of the great initial work from #728 based on the core refactor this was merged. Please everyone test that PR and report on there your results, so we can get this work in!

auipga, 2023-06-21:
@petejohanson thank you for putting effort in this issue.

**Test result 1**: The right encoder doesn't seem to respond yet.
Do you spot a mistake on my side? [my testing branch](https://github.com/auipga/zmk-config-hillside52/tree/test/split-encode)

```ini
sensor-bindings = <&inc_dec_kp C_VOL_DN C_VOL_UP &inc_dec_kp C_VOL_DN C_VOL_UP>;
```

<details>
<summary>west.yml</summary>

```yaml
manifest:
  remotes:
    - name: zmkfirmware
      url-base: https://github.com/zmkfirmware
    - name: stephen/split-encoder
      url-base: https://github.com/petejohanson
  projects:
    - name: zmk
      remote: stephen/split-encoder
      revision: main
      import: app/west.yml
  self:
    path: config
```
</details>

<details>
<summary>hillside52.conf</summary>

```ini
CONFIG_EC11=y
CONFIG_EC11_TRIGGER_GLOBAL_THREAD=y
```
</details>

petejohanson, 2023-06-21:
> @petejohanson thank you for putting effort in this issue.
>
> **Test result 1**: The right encoder doesn't seem to respond yet.
> Do you spot a mistake on my side? [my testing [...]

Can you enable logging and get the log results from both sides?

auipga, 2023-06-21:
> Can you enable logging and get the log results from both sides?

I followed https://zmk.dev/docs/development/usb-logging and added `CONFIG_ZMK_USB_LOGGING=y`. The filesize increased a lot therefore I thought it should work.
<details>
<summary>$ sudo tio /dev/ttyACM0</summary>

```
[16:33:40.050] tio v2.6
[16:33:40.050] Press ctrl-t q to quit
[16:33:40.050] Warning: Could not open tty device (No such file or directory)
[16:33:40.050] Waiting for tty device..

## Waits indefinitely
```

</details>
<details>
<summary>$ sudo dmesg -HW</summary>

```
[Jun25 15:21] usb 3-2.2: new full-speed USB device number 50 using xhci_hcd
[  +0,093013] usb 3-2.2: New USB device found, idVendor=1d50, idProduct=615e, bcdDevice= 3.00
[  +0,000005] usb 3-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0,000002] usb 3-2.2: Product: Hillside52
[  +0,000001] usb 3-2.2: Manufacturer: ZMK Project
[  +0,000001] usb 3-2.2: SerialNumber: 6D0E84BC8F8948EB
[  +0,011819] input: ZMK Project Hillside52 Keyboard as /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb3/3-2/3-2.2/3-2.2:1.2/0003:1D50:615E.0031/input/input110
[  +0,055355] hid-generic 0003:1D50:615E.0031: input,hidraw4: USB HID v1.10 Keyboard [ZMK Project Hillside52] on usb-0000:39:00.0-2.2/input2
```
</details>

I disabled bluetooth and used `&out OUT_USB` to ensure all data going via USB.

auipga, 2023-06-21:
> Can you enable logging and get the log results from both sides?

After a reboot it worked.
I ran the command, waited a few seconds, rotated left, waited a few seconds, rotated right
<details>
<summary>sudo tio /dev/ttyACM0 # left half</summary>

```
[17:03:50.289] tio v2.6
[17:03:50.289] Press ctrl-t q to quit
[17:03:50.290] Connected
c: data: 0
[00:29:59.834,777] <dbg> zmk: split_central_notify_func: data: 0
[00:29:59.834,777] <dbg> zmk: split_central_notify_func: data: 0
[00:29:59.834,777] <dbg> zmk: split_central_notify_func: data: 0
[00:29:59.834,808] <dbg> zmk: split_central_notify_func: data: 0
[00:29:59.834,808] <dbg> zmk: split_central_notify_func: data: 0
[00:29:59.834,838] <dbg> zmk: split_central_notify_func: data: 0
[00:29:59.834,838] <dbg> zmk: split_central_notify_func: data: 0
[00:29:59.834,838] <dbg> zmk: split_central_notify_func: data: 0
[00:29:59.834,869] <dbg> zmk: split_central_notify_func: data: 0
[00:29:59.834,869] <dbg> zmk: split_central_notify_func: data: 0
[00:29:59.834,899] <dbg> zmk: split_central_notify_func: data: 0
[00:29:59.834,899] <dbg> zmk: split_central_notify_func: data: 0
[00:29:59.834,899] <dbg> zmk: split_central_notify_func: data: 0
[00:29:59.834,930] <dbg> zmk: split_central_notify_func: data: 0
[00:29:59.834,991] <dbg> zmk: peripheral_event_work_callback: Trigger key position state change for 45
[00:29:59.835,021] <dbg> zmk: position_state_changed_listener: 45 bubble undecided hold-tap keyrelease event
[00:29:59.835,052] <dbg> zmk: tap_dance_position_state_changed_listener: Ignore upstroke at position 45.
[00:29:59.835,083] <dbg> zmk: zmk_keymap_apply_position_state: layer: 0 position: 45, binding name: LAYER_TAP
[00:29:59.835,174] <dbg> zmk: decide_hold_tap: 45 decided tap (tap-preferred decision moment key-up)
[00:29:59.835,174] <dbg> zmk: on_keymap_binding_pressed: position 45 keycode 0x70028
[00:29:59.835,205] <dbg> zmk: hid_listener_keycode_pressed: usage_page 0x07 keycode 0x28 implicit_mods 0x00 explicit_mods 0x00
[00:29:59.835,235] <dbg> zmk: zmk_hid_implicit_modifiers_press: Modifiers set to 0x00
[00:29:59.835,235] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[00:29:59.835,327] <dbg> zmk: on_keymap_binding_released: position 45 keycode 0x70028
[00:29:59.835,357] <dbg> zmk: hid_listener_keycode_released: usage_page 0x07 keycode 0x28 implicit_mods 0x00 explicit_mods 0x00
[00:29:59.835,357] <dbg> zmk: zmk_hid_implicit_modifiers_release: Modifiers set to 0x00
[00:29:59.835,388] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[00:29:59.835,937] <dbg> zmk: on_hold_tap_binding_released: 45 cleaning up hold-tap
[00:30:00.625,915] <dbg> zmk: vddh_sample_fetch: ADC raw 1394 ~ 5100 mV => 100%

## Left encoder turn left:

[00:30:14.518,310] <dbg> zmk: zmk_sensors_trigger_handler: sensor 0
[00:30:14.518,371] <dbg> zmk: zmk_keymap_sensor_triggered: layer: 0 sensor_number: 0, binding name: ENC_KEY_PRESS
[00:30:14.518,432] <dbg> zmk: on_sensor_binding_triggered: inc keycode 0xC00EA dec keycode 0xC00E9
[00:30:14.518,432] <dbg> zmk: zmk_keymap_sensor_triggered: Behavior returned error: -134
[00:30:14.518,463] <dbg> zmk: zmk_event_manager_handle_from: Listener returned an error: -134
[00:30:14.667,724] <dbg> zmk: zmk_sensors_trigger_handler: sensor 0
[00:30:14.667,816] <dbg> zmk: zmk_keymap_sensor_triggered: layer: 0 sensor_number: 0, binding name: ENC_KEY_PRESS
[00:30:14.667,846] <dbg> zmk: on_sensor_binding_triggered: inc keycode 0xC00EA dec keycode 0xC00E9
[00:30:14.667,877] <dbg> zmk: on_sensor_binding_triggered: SEND 786666
[00:30:14.667,907] <dbg> zmk: hid_listener_keycode_pressed: usage_page 0x0C keycode 0xEA implicit_mods 0x00 explicit_mods 0x00
[00:30:14.667,907] <dbg> zmk: zmk_hid_implicit_modifiers_press: Modifiers set to 0x00
[00:30:14.667,938] <dbg> zmk: zmk_endpoints_send_report: usage page 0x0C
[00:30:14.673,065] <dbg> zmk: hid_listener_keycode_released: usage_page 0x0C keycode 0xEA implicit_mods 0x00 explicit_mods 0x00
[00:30:14.673,095] <dbg> zmk: zmk_hid_implicit_modifiers_release: Modifiers set to 0x00
[00:30:14.673,095] <dbg> zmk: zmk_endpoints_send_report: usage page 0x0C
[00:30:14.868,072] <dbg> zmk: zmk_sensors_trigger_handler: sensor 0
[00:30:14.868,133] <dbg> zmk: zmk_keymap_sensor_triggered: layer: 0 sensor_number: 0, binding name: ENC_KEY_PRESS
[00:30:14.868,194] <dbg> zmk: on_sensor_binding_triggered: inc keycode 0xC00EA dec keycode 0xC00E9
[00:30:14.868,194] <dbg> zmk: zmk_keymap_sensor_triggered: Behavior returned error: -134
[00:30:14.868,225] <dbg> zmk: zmk_event_manager_handle_from: Listener returned an error: -134
[00:30:15.157,440] <dbg> zmk: zmk_sensors_trigger_handler: sensor 0
[00:30:15.157,531] <dbg> zmk: zmk_keymap_sensor_triggered: layer: 0 sensor_number: 0, binding name: ENC_KEY_PRESS
[00:30:15.157,592] <dbg> zmk: on_sensor_binding_triggered: inc keycode 0xC00EA dec keycode 0xC00E9
[00:30:15.157,592] <dbg> zmk: zmk_keymap_sensor_triggered: Behavior returned error: -134
[00:30:15.157,592] <dbg> zmk: zmk_event_manager_handle_from: Listener returned an error: -134

[... 36 lines cut ...]
[00:44:53.858,184] <dbg> zmk: zmk_usb_get_conn_state: state: 6
[00:44:53.981,475] <dbg> zmk: zmk_usb_get_conn_state: state: 1
[00:44:54.132,843] <dbg> zmk: zmk_usb_get_conn_state: state: 3
[00:45:00.359,466] <dbg> zmk: vddh_sample_fetch: ADC raw 1388 ~ 5080 mV => 100%
[00:45:00.359,497] <dbg> zmk: zmk_battery_update: Setting BAS GATT battery level to 100.

## Right encoder turn left

[... cut ...]
[00:45:18.362,335] <dbg> zmk: zmk_sensors_trigger_handler: sensor 1

## Right encoder turn right

[00:45:28.667,663] <dbg> zmk: zmk_sensors_trigger_handler: sensor 1
[00:45:28.702,270] <dbg> zmk: zmk_sensors_trigger_handler: sensor 1
[00:45:28.901,153] <dbg> zmk: zmk_sensors_trigger_handler: sensor 1
[00:45:29.101,287] <dbg> zmk: zmk_sensors_trigger_handler: sensor 1
[00:45:29.102,142] <dbg> zmk: zmk_sensors_trigger_handler: sensor 1
[00:45:29.102,447] <dbg> zmk: zmk_sensors_trigger_handler: sensor 1

[17:07:07.637] Disconnected
```
</details>

caksoylar, 2023-06-21:
> ```yaml
>
> manifest:
>
>   remotes:
>
>     - name: zmkfirmware
>
>       url-base: https://github.com/zmkfirmware
>
>     - name: stephen/split-encoder
>
>       url-base: https://github.com/petejohanson [...]

If I am not wrong, this is referring to Pete's main branch, not the PR branch.

petejohanson, 2023-06-21:
> If I am not wrong, this is referring to Pete's main branch, not the PR branch.

Yeah, that's definitely not right.

auipga, 2023-06-26:
> If I am not wrong, this is referring to Pete's main branch, not the PR branch.

Thanks. It works with the right branch.

<details>
<summary>west.yml (updated)</summary>

```yaml
manifest:
  remotes:
    - name: zmkfirmware
      url-base: https://github.com/zmkfirmware
    - name: petejohanson
      url-base: https://github.com/petejohanson
  projects:
    - name: zmk
      remote: petejohanson
      revision: stephen/split-encoder
      import: app/west.yml
  self:
    path: config
```
</details>

Now playing with the `steps` and `triggers-per-rotation` as mentioned in this [blog post](https://zmk.dev/blog/2023/06/18/encoder-refactors)

petejohanson, 2023-06-26:
> As I don't use ec11 but this [ec12](https://keycapsss.com/keyboard-parts/parts/220/alps-alpine-encoder-low-profile-ec12) and I don't know about the differences in this regard I hope it's still applicable.
> 
> The encoder I use has **12** detents and it used to work well with `resolution` of **4**.
> My guess would be that `steps` should be 12x4=**48** and `triggers-per-rotation` = **12**.
> To test it I assigned <kbd><</kbd> and <kbd>></kbd>.
> 
> These are the results of 12 'clicks', separated by a new line:
> <details>
> <summary>left half, clockwise:</summary>
> 
> ```
> ><><
> ><><
> ><><
> ><><
> ><><
> ><><
> ><<
> ><><
> ><<
> ><><
> ><<
> ><<
> ```
> </details>
> <details>
> <summary>left half, anti-clockwise:</summary>
> 
> ```
> <><>
> ><>
> <><>
> <><>
> <><>
> <><>
> <>>
> <><>
> <><>
> <><>
> <><>
> <><>
> ```
> </details>
> <details>
> <summary>right half, clockwise:</summary>
> 
> ```
> <><>
> <><>
> <>>
> <><>
> <><>
> <><>
> <><>
> <>><
> <>>
> ><>
> <><>
> <><>
> ```
> </details>
> <details>
> <summary>right half, anti-clockwise:</summary>
> 
> ```
> ><><
> ><><
> ><<
> ><><
> ><><<
> <><
> >><><
> ><><
> ><><
> ><><
> ><><
> ><><
> ```
> </details>
> 

That should, in theory, work. Can you get the logs for each scenario as well?

auipga, 2023-06-26:
<details>
<summary>sudo tio /dev/ttyACM0</summary>

```
[18:31:09.642] tio v2.6
[18:31:09.642] Press ctrl-t q to quit
[18:31:09.643] Connected
full, drain buffer
--- 118 messages dropped ---
[02:08:15.874,023] <inf> usb_cdc_acm: Ring buffer full, drain buffer
--- 118 messages dropped ---
[02:08:15.880,004] <inf> usb_cdc_acm: Ring buffer full, drain buffer
--- 113 messages dropped ---
[02:08:15.886,291] <inf> usb_cdc_acm: Ring buffer full, drain buffer
--- 118 messages dropped ---
[02:08:15.892,974] <inf> usb_cdc_acm: Ring buffer full, drain buffer
--- 127 messages dropped ---
[02:08:15.899,200] <inf> usb_cdc_acm: Ring buffer full, drain buffer
--- 118 messages dropped ---
[02:08:15.905,334] <inf> usb_cdc_acm: Ring buffer full, drain buffer
--- 127 messages dropped ---
[02:08:15.911,926] <inf> usb_cdc_acm: Ring buffer full, drain buffer
--- 115 messages dropped ---
[02:08:15.918,182] <inf> usb_cdc_acm: Ring buffer full, drain buffer
--- 118 messages dropped ---
[02:08:15.924,438] <inf> usb_cdc_acm: Ring buffer full, drain buffer
--- 48 messages dropped ---
[02:08:15.925,323] <dbg> zmk: zmk_hid_implicit_modifiers_press: Modifiers set to 0x00
--- 8 messages dropped ---
[02:08:15.925,354] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[02:08:15.925,415] <dbg> zmk: behavior_queue_process_next: Processing next queued behavior in 30ms
[02:08:15.925,567] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.925,598] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.925,659] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.925,720] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.925,750] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.925,811] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.925,872] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.925,903] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.925,964] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.926,025] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.926,055] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.926,116] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.926,177] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.926,208] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.926,269] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.926,330] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.926,361] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.926,422] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.926,483] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.926,544] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.926,574] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.926,635] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.926,696] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.926,727] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.926,788] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.926,849] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.926,879] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.926,940] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.927,001] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.927,032] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.927,093] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.927,154] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.927,215] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.927,276] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.927,307] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.927,368] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.927,398] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.927,459] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[02:08:15.927,520] <inf> usb_cdc_acm: Ring buffer full, drain buffer
[... 1057 lines cut ...]
[02:08:57.844,604] <dbg> zmk: zmk_behavior_sensor_rotate_common_process: Sensor binding: ENC_KEY_PRESS
[02:08:57.844,665] <dbg> zmk: zmk_keymap_sensor_event: sensor event processing complete, behavior response was opaque
[02:08:57.848,144] <dbg> zmk: behavior_queue_process_next: Invoking KEY_PRESS: 0x2070064 0x00
[02:08:57.848,175] <dbg> zmk: on_keymap_binding_released: position 0 keycode 0x2070064
[02:08:57.848,205] <dbg> zmk: hid_listener_keycode_released: usage_page 0x07 keycode 0x64 implicit_mods 0x02 explicit_mods 0x00
[02:08:57.848,236] <dbg> zmk: zmk_hid_implicit_modifiers_release: Modifiers set to 0x00
[02:08:57.848,236] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[02:08:57.848,327] <dbg> zmk: behavior_queue_process_next: Processing next queued behavior in 0ms
[02:08:57.848,358] <dbg> zmk: behavior_queue_process_next: Invoking KEY_PRESS: 0x70064 0x00
[02:08:57.848,419] <dbg> zmk: on_keymap_binding_pressed: position 0 keycode 0x70064
[02:08:57.848,449] <dbg> zmk: hid_listener_keycode_pressed: usage_page 0x07 keycode 0x64 implicit_mods 0x00 explicit_mods 0x00
[02:08:57.848,480] <dbg> zmk: zmk_hid_implicit_modifiers_press: Modifiers set to 0x00
[02:08:57.848,480] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[02:08:57.848,876] <dbg> zmk: behavior_queue_process_next: Processing next queued behavior in 5ms
[... cut ...]
[02:08:57.854,125] <dbg> zmk: hid_listener_keycode_released: usage_page 0x07 keycode 0x64 implicit_mods 0x00 explicit_mods 0x00
[02:08:57.854,125] <dbg> zmk: zmk_hid_implicit_modifiers_release: Modifiers set to 0x00
[02:08:57.854,156] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[02:08:57.854,217] <dbg> zmk: behavior_queue_process_next: Processing next queued behavior in 0ms
[02:09:00.609,039] <dbg> zmk: vddh_sample_fetch: ADC raw 1404 ~ 5140 mV => 100%

[18:31:57.187] Disconnected
```
</details>

auipga, 2023-06-27:
**Result 3:** The previous misbehaviour of ~4 keystrokes isn't present anymore.
(I didn't detect the decisive difference. I'm using my main branch now.)

What I experience now is that when I `&lt` or `&tog` to another layer, I get keystrokes for the default layer aswell as the keystrokes from the momentary layer.
I'll do some more testing and let you know.

auipga, 2023-06-27:
**Result 4:**
- misbehaviour of ~4 keystrokes is there as soon as the last layer has sensor-bindings (all 4 set to SPACE in my case). See [this](https://github.com/auipga/zmk-config-hillside52/commit/c55a006ecebfaacf3ef5f88910a159e515387f0e) commit, which does only comment it out and after that commit the steps/resolution thing works correctly.
- Multiple keypresses with layers: Each momentary layer that is involved sends its keys. My default layer sends volume controls, the next &lt layer sends LEFT/RIGHT, the next &lt sub-layer sends UP/DOWN. Rotation results in diagonal cursor movement and volume change at the same time. An almost identical sublayer seems to behave a little different. The mentioned keypresses still occur but they are not at the same exact degree. Turning the knob slowly (given 4 steps per trigger) shows the keypresses one after another. It just makes no sense to me.

auipga, 2023-07-07:
**Result 5:** SUCCESS
- `settings_reset` for the win. All mentioned misbehaviours are gone. ❤️
- other problems I'm experiencing are also present without your feature branch and are therefore unrelated.

Thank you. It's soooo good. Using my [Hillside52](https://github.com/mmccoyd/hillside) is much more productive now. Well done @petejohanson. You made me very happy!

If you don't require other users to test, this issue can be closed from my side.

closed 2023-07-07 by auipga